qmk/qmk_firmware #756: tapdance does not work.

benji877, 2016-09-14:
By adding this line (#define KC_TD TD) I got it to compile, but if I double tap 1 (I want it to type !) it switches to layer one and stays there.

Keymap: http://pastebin.com/hGArVhhS

algernon, 2016-09-14:
Do you have `TAP_DANCE_ENABLE=yes` in your `Makefile`? Does single tapping work?

benji877, 2016-09-14:
Yes and yes.

algernon, 2016-09-14:
Hm. Can you try something for me? If you make a macro (and bind it to a key, of course) that does the following, does that input `!`, or does it switch to layer one?

```c
case M_TEST:
  if (record->event.pressed) {
    register_code16(KC_EXLM);
  } else {
    unregister_code16(KC_EXLM);
  }
  break
```

algernon, 2016-09-14:
Also, does double-tapping `2` work? Or does that switch to layer two?

benji877, 2016-09-14:
Double tapping 2 puts me into layer 2.

algernon, 2016-09-14:
Okay. Can you try the macro too? That will give me a general idea where to look for bugs.

Thanks!

benji877, 2016-09-14:
I will try it when I get home.

algernon, 2016-09-23:
@benji877 did you have a chance to try the macro? Unfortunately, I'm still unable to reproduce the problem :(

benji877, 2016-09-23:
Oops. I forgot. Yes, I will try to do it this weekend.

closed 2017-06-27 by skullydazed