qmk/qmk_firmware #759: Should we use the Wiki more?

DidierLoiseau, 2016-09-17:
The QMK documentation seems to be a bit scattered here and there:

* there is a **huge readme** at the root of the repository, that is difficult to navigate on GitHub (a bit better on [qmk.fm](http://qmk.fm)) and probably provides a bit too much of information at once for newcomers;
* this readme references several **additional files inside the `doc` directory** and elsewhere – internal links do not work on qmk.fm though;
* there are also some **references to code files** (mainly header files I guess);
* there is a **wiki**, however it does not seem to have been maintained much since the TMK fork and it lacks structure and content;
* there are still **links to TMK**, and a lot of them are probably no more relevant – I think TMK deserves to be mentioned, however it is an historical detail that is irrelevant for newcomers. The documentation of the project itself should be self-contained.

There are several issues with the documentation when it is put inside the sources:
* each change **requires a new commit**;
* contributors must make **pull requests** when they want to improve the documentation;
* the two points above probably **make people reluctant** from contributing to the documentation, especially for small changes;
* **splitting documentation** across several files **is difficult**, but maintaining long pages is not really better;
* **links can easily get broken**, and broken links are difficult to spot – on the wikis, [broken internal links are displayed in red](https://help.github.com/articles/adding-links-to-wikis/#using-a-wiki-markup-syntax-to-make-links);
* we don't have a **centralised view** of the documentation.

For me, the readme's should only be used to:
* provide a **description** of the project (at the root of the repository);
* describe the **contents of the directory** it sits in (e.g. keyboards, keymaps etc.);
* provide **pointers to useful resources**.

I think outside of this, there shouldn't be any documentation files in the repository – just documentation/comments inside the code itself.

See also [About GitHub Wikis](https://help.github.com/articles/about-github-wikis/):

> GitHub Wikis are a place in your repository where you can share long-form content about your project, such as how to use it, how it's been designed, manifestos on its core principles, and so on. Whereas a README is intended to quickly orient readers as to what your project can do, wikis can be used to provide additional documentation.

So, what do you think about migrating the documentation to Wiki pages?

ezuk, 2016-09-18:
I for one love the idea!

algernon, 2016-09-18:
For added fun, the WiKis on GitHub are also cloneable repos, so one can build a website out of them too, with different formatting and whatnot.

jackhumbert, 2016-09-19:
Definitely something that needs done! Thanks for bringing it up :)

DidierLoiseau, 2016-09-19:
I guess if everyone agrees, we must simply determine the way forward :smiley: 

I propose to proceed as follows:

- [ ] Migrate the current documentation files from the repo to the wiki (trying to keep the links between pages if possible);
  - [ ] Copy the contents to the wiki (maybe keeping the same structure, or put everything)
  - [ ] Remove the files from the repo
- [ ] Choose an appropriate structure for the wiki pages (discuss here) – we will probably need to [create a sidebar](https://help.github.com/articles/creating-a-sidebar/);
- [ ] Reorganise the wiki pages accordingly.

@jackhumbert For the migration, would you prefer to do it yourself, or someone could already copy the contents and possibly even make a PR to remove the files?

For reference, I just came across the [Projects with great wikis](https://github.com/showcases/projects-with-great-wikis) page, which could serve as examples.

fredizzimo, 2016-12-06:
I think the current situation is a little bit confusing.

The main documentation was copied to the wiki, but the problem is that it keeps getting updated, and the wiki gets updated with other things, and we are having trouble keeping them both in sync.

So should we just change the main readme.md file to link to the wiki, and doucment all features there? 

Another problem is that new features should ideally have documentation with them, but if we use the Wiki there's a chicken/egg problem, as it's not possible (as I'm aware of) to send pull request to the wiki itself. So that means that we have to edit the wiki before the feature has been accepted, and that might confuse someone reading it at that moment. Do you have any suggestion of how to handle that?

jackhumbert, 2016-12-06:
The lack of PRs on the wiki make conversion odd, but I think anyone can edit the wiki currently (please correct me if I'm wrong) - we just need to copy things over a final time, correct all the links, and edit the readme to point to the wiki. The last part there is the only thing that can be in a PR.

Rather than typing this out for someone else to get to, I'll just do it now haha.

Regarding the timing of PRs/wiki changes, I'd say we just stick in a flag like [PROPOSED CHANGE] until it gets merged. As long as the part in the wiki that's been edited is linked in the PR, it should be easy to maintain.

fredizzimo, 2016-12-06:
Yes, that makes sense. And also in many cases we can included the proposed wiki changes, at least for new features where we just add some new sections, in the pull request description. That can then easily be copied to the wiki once it's merged.

jackhumbert, 2016-12-06:
Yeah, it all being markdown makes that easy.

The main readme has been copied and stripped. Link editing/organising can now be done :) that includes removing the some of the doc files, but that can be done via PR since they aren't viewed as often.

fredizzimo, 2016-12-06:
Another thing, I just noticed it's possible to clone the wiki `https://github.com/jackhumbert/qmk_firmware.wiki.git`, so it should be possible to clone it locally and edit it there. You could then share it for review on your own github account. You would have to push the changes to QMK though, rather than send a pull request.

We could even have a clone of the wiki hosted on your account, where we can send pull request, then you could merge them like normal and push them to the official wiki. 

It's a quite complicated process, so I don't recommend it for everyone, but it's certainly possible.

jackhumbert, 2016-12-06:
Yeah, I've messed around with that a little - I don't think it's possible to share/fork it for review, though. When pushing, it goes straight to the qmk wiki. Is there something I'm missing there?

fredizzimo, 2016-12-06:
I was thinking about having a separate repository where it's reviewed. In the first case I tried to explain it would be the submitter's own repository and in the second case a repository owned by you.

In the first case it's up to rely on the submitter to only push when he has got an OK, and in the second case it's more work for you. You would first need to merge the pull request, and then push it to `qmk_firmware.wiki.git`

jackhumbert, 2016-12-06:
Haha, [awesome](https://github.com/jackhumbert/qmk_firmware_wiki). It'd be nice to have hooks set up to pull into that one, and push into the wiki here, and [it looks like the api supports that event on the wiki side of things](https://developer.github.com/v3/activity/events/types/#gollumevent).

kekumu, 2017-02-26:
When I use a wiki for a project, I usually add the wiki's repo as a submodule at `docs/`. I find it to be a nice way to keep documentation together with the code, but also have it editable through Github's wiki tools in the browser.

jackhumbert, 2017-02-26:
Ah, that's a good idea!

fredizzimo, 2017-03-31:
I wonder if the keyboard and keymap readme files should be moved to the wiki as well? I know that there's the [qmk.fm](http://qmk.fm/) generated site, but that could probably be constructed inside the wiki.

Also many of the keyboard readme files have some flashing information for example, that would be better linked to some more general wiki page, as the same information is repeated everywhere. I think that would be easier to handle if everything were in the same wiki.

The disadvantage is that it's probably slightly more work for the keyboard and keymap creators, since you need to add the wiki page to the right structure. It can also easily get out of sync if keyboards/keymaps are deleted or renamed.

I don't really have a personal preference for this, I'm just adding this as a thought.

algernon, 2017-03-31:
I would be against moving keymap readmes to the WiKi, because the primary repo of my keymap is just the keymap, without the rest of QMK. Currently, merging new versions of my keymap to QMK is easy, I just copy the files over, squash the git log, and edit it by hand a little.

Would the README move to the WiKi, I'd have a much harder time, and likely wouldn't update the wiki at all. Too much work.

skullydazed, 2017-06-27:
Closing this issue because we have moved to the source tree and are now using gitbook to generate http://docs.qmk.fm.

closed 2017-06-27 by skullydazed