qmk/qmk_firmware #762: Review the Eclipse wiki page

DidierLoiseau, 2016-09-18:
Hello,

I just created a [Wiki page](../wiki/Eclipse) describing the setup of the Eclipse IDE for QMK.

I would like to have some feedback on whether it works with other environments, as well as possible improvements.

Would anyone mind to review it?

Thanks in advance :smiley:

IBNobody, 2016-09-20:
Were you able to get debugging to work (assuming you had an AVR debugger)?

DidierLoiseau, 2016-09-20:
@IBNobody I didn't try. I have used Eclipse in the past to debug C programs but this was 8-10 years ago.

I think it requires at least to compile with debug symbols, which I suppose would be sufficient for debugging unit tests. I don't know whether debugging the Teensy 2 is actually possible.

IBNobody, 2016-09-20:
@DidierLoiseau Ok, thanks. I wasn't sure. I have Atmel Studio, and I should be able to debug the Teensy with my JTAG debugger.

DidierLoiseau, 2016-09-20:
@IBNobody I just had a look at JTAG, but it seems it requires physical access to the board, probably even soldering directly on the CPU, right?

IBNobody, 2016-09-20:
@DidierLoiseau The JTAG pins are multiplexed with some of the F port pins, but they are all brought out to pins on a Teensy 2.0. You lose keyboard functionality when you debug this way, though. It is meant for bootloader debugging.

fredizzimo, 2016-09-25:
I have also been using eclipse with QMK. One thing that is not described is how to configure it for the unit test support (including running the debugger for them), so I will add that to the description at some point. I'm using it in combination with Cygwin though, so the process would be slightly different for mingw.

DidierLoiseau, 2016-09-26:
@fredizzimo I haven't tried running unit tests yet, so don't hesitate to describe it! :smiley:

Another thing that is still missing, because I didn't dive into it either, is how to get Eclipse to understand all `#define`'d constants. By default a lot of them are already recognised, but for some it seems it requires some tuning (e.g. `MATRIX_ROWS`/`COLS` which are always shown in error). I think it gets confused when there are several definitions.

jackhumbert, 2017-06-27:
I think everything here's been mostly resolved (it's since been moved [here](https://github.com/qmk/qmk_firmware/blob/master/docs/eclipse.md)) - @DidierLoiseau, if I'm mistaken, please reopen this issue!

closed 2017-06-27 by jackhumbert

fredizzimo, 2017-06-27:
The page is not included on [docs.qmk.fm](https://docs.qmk.fm)

jackhumbert, 2017-06-27:
Fixed at 4ba9438 :)