qmk/qmk_firmware #766: Add support for avrdude

thatismatt, 2016-09-20:
I've started work on this on my fork: thatismatt/qmk_firmware@8af4ab76b620b719a1807f3b7392fe8670496116

Usage would require adding the following to a keyboards rules.mk
```
AVRDUDE_PROGRAMMER = avr109
AVRDUDE_PORT       = /dev/ttyACM0
```

And then just:
```
make avrdude
```

I have verified that this works for me with a pro micro.

Mostly this was taken from the atreus makefiles (https://github.com/jackhumbert/qmk_firmware/blob/master/keyboards/atreus/rules.mk#L10).

I'd welcome any feedback to turn this in to something that you'd be happy to merge.

IBNobody, 2016-09-20:
Why avrdude and not dfu-programmer? Just curious.

thatismatt, 2016-09-21:
Good question, probably familiarity. I've no experience with the dfu-programmer. I'm coming at this having a bit of experience programming AVRs with a USBasp.

What are the pros/cons for using the dfu-programmer vs avrdude?

IBNobody, 2016-09-21:
dfu-programmer programs all of the USB equipped AVRs. The builds of our toolchain already include it. There is nothing that needs to be configured. It doesn't require a dongle. It is cross-platform. Making is as easy as `make dfu` or `make planck-default-dfu`. 

So for our application, the two programmers may be identical (except that avrdude is not supported yet).

thatismatt, 2016-09-21:
Thanks that is very useful.

So there is no requirement for a particular bootloader? i.e. will it work on a Pro Micro with the Arduino Caterina bootloader?

skullydazed, 2016-09-21:
If you're using a ProMicro you will need to use avrdude. You can't use dfu-programmer.

IBNobody, 2016-09-21:
The bootloader has to support the DFU format of USB programming. I took a peek at Caterina, and I saw that it uses a CDC virtual serial port (boo). So, no, they are not compatible. This is why I was asking. :)

Go ahead and add it similar to how the Teensy CLI is implemented. And if you can swing it, try and get it to work on a Windows 7 box.

In researching this, I ran into a parallel question, which I will post.

IBNobody, 2016-09-21:
Also, FYI: If you start to run out of flash space (very possible with larger keyboards, layers, and audio), get a Teensy. It has a proprietary bootloader that is 8x smaller than DFU or Caterina. You'll recover 3.5k of space!

@jackhumbert I am setting my Atmel debugger up soon to see if I can shrink the DFU bootloader. That would give us the biggest savings in flash space for chip-based keyboards like the Planck and Atomic.

IBNobody, 2016-09-29:
@thatismatt 
It looks like AVRDUDE is part of the Windows MHV AVR toolchain, which is what the documentation recommends that Windows users use. So it's there if you need it...

```
Filename = avrdude.exe  
Location = C:\Program Files (x86)\MHV AVR Tools\bin\avrdude.exe
```

What about other OS users?

thatismatt, 2016-10-07:
I'm on Linux, and don't have easy access to test on a Windows box.

If it is useful I can create a pull request from my branch with the code I'm using that works under Linux.

kajsa, 2017-10-18:
I can test on Windows 7 (coming to this late, though)

drashna, 2018-03-25:
This has been added, and confirmed working since #2301

closed 2018-03-25 by drashna