qmk/qmk_firmware #770: Are the AVR Bootloader images correct?

IBNobody, 2016-09-21:
Take a look at this from the bootloader.c file...

```
/* Bootloader Size in *bytes*
 *
 * AVR Boot section size are defined by setting BOOTSZ fuse in fact. Consult with your MCU datasheet.
 * Note that 'Word'(2 bytes) size and address are used in datasheet while TMK uses 'Byte'.
 *
 *
 * Size of Bootloaders in bytes:
 *   Atmel DFU loader(ATmega32U4)   4096
 *   Atmel DFU loader(AT90USB128)   8192
 *   LUFA bootloader(ATmega32U4)    4096
 *   Arduino Caterina(ATmega32U4)   4096
 *   USBaspLoader(ATmega***)        2048
 *   Teensy   halfKay(ATmega32U4)   512
 *   Teensy++ halfKay(AT90USB128)   1024
 *
 *
 * AVR Boot section is located at the end of Flash memory like the followings.
 *
 *
 * byte     Atmel/LUFA(ATMega32u4)          byte     Atmel(AT90SUB128)
 * 0x0000   +---------------+               0x00000  +---------------+
 *          |               |                        |               |
 *          |               |                        |               |
 *          |  Application  |                        |  Application  |
 *          |               |                        |               |
 *          =               =                        =               =
 *          |               | 32KB-4KB               |               | 128KB-8KB
 * 0x6000   +---------------+               0x1FC00  +---------------+
 *          |  Bootloader   | 4KB                    |  Bootloader   | 8KB
 * 0x7FFF   +---------------+               0x1FFFF  +---------------+
 *
 *
 * byte     Teensy(ATMega32u4)              byte     Teensy++(AT90SUB128)
 * 0x0000   +---------------+               0x00000  +---------------+
 *          |               |                        |               |
 *          |               |                        |               |
 *          |  Application  |                        |  Application  |
 *          |               |                        |               |
 *          =               =                        =               =
 *          |               | 32KB-512B              |               | 128KB-1KB
 * 0x7E00   +---------------+               0x1FC00  +---------------+
 *          |  Bootloader   | 512B                   |  Bootloader   | 1KB
 * 0x7FFF   +---------------+               0x1FFFF  +---------------+
 */
```

I spotted an error and wanted to confirm. 

The Atmel/LUFA(ATMega32u4) chart shows that the application range is from 0x0000 to 0x5FFF. However, math shows that it should be 0x0000 to **0x6FFF**. Also, the AT90's should go from 0x00000 to **0x1DFFF**. The Teensy ones look correct.

shelaf, 2016-09-22:
Probably, you are right.

[Atmel DFU loader]
See _Table2-1_
Pay attention to word address.
http://www.atmel.com/Images/doc7618.pdf

[Teensy, Teensy++]
See _Jump address_
https://www.pjrc.com/teensy/jump_to_bootloader.html

fredizzimo, 2016-09-25:
Yes, it looks like you are right. But as far as I can see the code still calculates things correctly, so only the comments are wrong.

IBNobody, 2016-09-26:
Yes, only the comments are wrong.

commit 78767bf2 (2016-09-29): Fix #770, Fix #779
commit f0db230c (2016-09-29): Merge pull request #790 from IBNobody/master

Fix #770, Fix #779, Update Vision
closed 2016-09-29 by jackhumbert