qmk/qmk_firmware #775: Layer display on LED segment

tobsn, 2016-09-22:
Would it be possible with a teensy/teensy++ to display the function layer on a LED segment?
This came to me last night after looking at the ergodox infinity... it would be fairly easy to put a LED segment block into common keyboard layouts and it would help a lot navigating the layers.

mecanogrh, 2016-09-23:
MAX7219

mecanogrh, 2016-09-23:
Then have a look at led control code from Eberhard Fahle

tobsn, 2016-09-23:
Fahle's LedControl code says "The code also works with the Teensy (3.1)"

mecanogrh, 2016-09-24:
Problem here is power consumption vs led brightness.

fredizzimo, 2016-09-25:
One goal of the [Effect System](https://github.com/jackhumbert/qmk_firmware/issues/717), is to be able to do that. The keyboard will need some sort of LED controller though. The Infinity uses the 31FL3731C, but that's for monochrome LEDs. The MAX7219 is another option. For RGB LEDs, you need something else. 

The effect system will most likely expose the LEDs as an abstract display with individually addressable pixlels, so you could use for example normal font drawing algorithms to display some characters on them.

mrlinuxfish, 2016-12-12:
I have modified an older version of the firmware to utilize a caps lock led as well as layer indication leds by putting this code in led.c and adding led.c to the sources: 
```
#include <avr/io.h>
#include "stdint.h"
#include "led.h"

// Caps Lock led settings (set to B0)
void led_set(uint8_t usb_led)
{
    // Using B7 Caps Lock LED
    if (usb_led & (1<<USB_LED_CAPS_LOCK))
    {
        // Output high.
        DDRB |= (1<<0);
        PORTB |= (1<<0);
    }
    else
    {
        // Output low.
        DDRB &= ~(1<<0);
        PORTB &= ~(1<<0);
    }
}

// Led layer toggle
void led_layer_set(uint32_t state) {
    DDRB |= (1<<7); //led 0 (B7)
    //dprint("Set LEDs for layer change\n");

    // Led for Layer 1
    if ((1<<1 & state) != 0) {
        //dprint("Layer 2: on\n");
        PORTB |= (1<<7);
    } else {
        //dprint("Layer 2: off\n");
        PORTB &= ~(1<<7);
    }
	
    DDRB |= (1<<3); //led 1 (B3)
    //dprint("Set LEDs for layer change\n");

    // Led for Layer 4
    if ((1<<4 & state) != 0) {
        //dprint("Layer 2: on\n");
        PORTB |= (1<<3);
    } else {
        //dprint("Layer 2: off\n");
        PORTB &= ~(1<<3);
    }
	
    DDRB |= (1<<2); //led 2 (B2)
    //dprint("Set LEDs for layer change\n");

    // Led for Layer 2
    if ((1<<2 & state) != 0) {
        //dprint("Layer 2: on\n");
        PORTB |= (1<<2);
    } else {
        //dprint("Layer 2: off\n");
        PORTB &= ~(1<<2);
    }
	
    DDRB |= (1<<1); //led 3 (B1)
    //dprint("Set LEDs for layer change\n");

    // Led for Layer 3
    if ((1<<3 & state) != 0) {
        //dprint("Layer 2: on\n");
        PORTB |= (1<<1);
    } else {
        //dprint("Layer 2: off\n");
        PORTB &= ~(1<<1);
    }
}

```

I actually don't know the exact terminology for adding source files, this is how my old makefile handled importing files:
```
# project specific files
SRC +=	extended_keymap_common.c \
	matrix_handwire.c \
	led.c \
	backlight.c \
	extended_keymap_dustin.c
	
CONFIG_H = config.h
```

drashna, 2018-03-25:
Ergodox does this by default now, with the option to do so with RGB underglow. 

https://github.com/qmk/qmk_firmware/blob/35b44ac6999f4eb5e556d5931dafef9a47c0f2e6/keyboards/ergodox_ez/keymaps/default/keymap.c#L208-L277

Closing.   Please reopen or open a new issue, if this is still an issue.

mecanogrh, 2018-06-21:
Yes please reopen, totally different issue that mrlinuxfish and you are considering.
We are talking about LED **SEGMENT** here not single LEDs.

drashna, 2018-10-21:
This should be possible, in theory.  Though it depends on the hardware.

The Helix board uses an LCD display and can display the current layer.
And I use layer indication via rgb LEDs on several boards (underglow and matrix).

drashna, 2019-03-29:
@mecanogrh just an update on this.  Yes, this should be possible, ESPECIALLY with the new LED Matrix feature.  The only caveat is that the Ergodox Infinity would need to be converted to this code.  

https://docs.qmk.fm/#/feature_led_matrix?id=custom-layer-effects

I'm using the equivalent for the RGB Matrix on my ergodox ez glow, and planck light.  So it is definitely possible with the newer code.

jackhumbert, 2020-03-13:
I think we can close this now - the implementation of this seems pretty straight-forward for personal cases. If anyone has code they can share, we can open a new issue/PR about how we can generalise it for a driver/etc in QMK.

closed 2020-03-13 by jackhumbert