qmk/qmk_firmware #780: action_tapping.c: build fails with DEBUG_ACTION enabled (suspicious code)

DidierLoiseau, 2016-09-23:
While trying to get proper debug in `action.c`, I figured I had to set the `DEBUG_ACTION` compiler flag to have any debug output from all the `action*.c` files. This can be done by putting the following in `rules.mk`:
```make
EXTRAFLAGS += -DDEBUG_ACTION
```
However, if you do this, the build will fail with the following error:
```make
./tmk_core/common/action_tapping.c: In function ‘process_tapping’:
./tmk_core/common/action_tapping.c:260:87: error: expected ‘;’ before ‘{’ token
                 if (!IS_NOEVENT(event)) debug("Tapping: other key just after tap.\n") {};
                                                                                       ^
Compiling: ./tmk_core/common/action_tapping.c                                                       [ERRORS]
```
Indeed, here is what we have in [`action_tapping.c:260`](https://github.com/jackhumbert/qmk_firmware/blob/e1fa9073c263d260fc0ec3c0d2d8a6a6f5aee42e/tmk_core/common/action_tapping.c#L260-L262):
```c
if (!IS_NOEVENT(event)) debug("Tapping: other key just after tap.\n") {};
process_record(keyp);
return true;
```
The curly braces have been added in commit 4d4f7684e684bec319f166121463a88cd4a62703 (PR #465). Before that, the `if` body was just a no-op (just the semicolon) when debug was disabled.

I think the goal is to conditionally call `debug` based on `IS_NOEVENT()`, the check is not intended to affect the `process_record()` that follows.

To avoid this issue in the future, and for better clarity, the curly braces should always be used, as it is the case for other similar checks at other places in the same file, e.g.:
```c
if (!IS_NOEVENT(event)) {
    debug("Tapping: key event while last tap(>0).\n");
}
process_record(keyp);
return true;
```

fredizzimo, 2016-09-25:
I agree (and no I didn't write the code, even if the repository claims that I did)

Since I have the unsquashed version of the commit in my own repository I too a closer look that the change in question, it had a description of "Modularity and GCC warning fixes", so the reason for the change was probably to get rid of warnings with the original code. The TMK repository has more warning enabled than QMK.

I guess it was warning about an empty if statement. But of course the fix is wrong as you noticed.  But I think that even your suggested way of writing would trigger the -Wempty-body warning.

I think the correct way to fix it, and keep all the warnings enabled, would be to do something like this https://github.com/tmk/tmk_core/pull/17, a pull request that I sent to TMK a long time ago, but that has never been merged. But on the other hand, if we don't intend to enable all the warnings.

This could of course even be further extended to something like `debug_if(condition, msg)`, where the condition evaluation is disabled completely if the debugging is turned off. Otherwise we are relying on the compiler being able to detect that IS_NOEVENT doesn't have any side effects and can be optimized away.

DidierLoiseau, 2016-09-25:
Will the compiler not completely optimise out the condition if it sees its evaluation has no side effect, and the body is empty?

Anyway, the most important is to fix the code. After that, we could talk about optimisations :-)

drashna, 2018-03-25:
This doesn't appear to error out anymore. 

I'm closing this.  If the issue is still there, reopen this or open a new issue.

closed 2018-03-25 by drashna