qmk/qmk_firmware #782: Make for Planck not working

paulbunyon, 2016-09-24:
Have a suggestion for what I should do next?

Windows 10 64 bit dell used vista and above. Got this error when trying to produce the check output file
QMK Windows Check Output v1.0
.......The system cannot find the batch label specified - ReportVersion
..

Ran Make and got 
C:\WINDOWS\system32>cd C:\qmk_firmware-master\qmk_firmware-master\keyboards\planck

C:\qmk_firmware-master\qmk_firmware-master\keyboards\planck>make
      0 [main] date 10860 open_stackdumpfile: Dumping stack trace to date.exe.stackdump
      0 [main] date 10892 handle_exceptions: Exception: STATUS_ACCESS_VIOLATION
    514 [main] date 10892 open_stackdumpfile: Dumping stack trace to date.exe.stackdump
Error: cmp command not found, please install diffutils
../../Makefile:474: recipe for target 'planck-allsp-allkm' failed
make: *** [planck-allsp-allkm] Error 1

C:\qmk_firmware-master\qmk_firmware-master\keyboards\planck>

robotmaxtron, 2016-09-25:
For what it's worth, things are correctly compiling for me in Linux for the Planck.

fredizzimo, 2016-09-25:
The symptoms are the same as this #709. In other words mingw is not properly installed, so date.exe crashes, and the diff command can not be found.

It would also be good if you could run @IBNobody's new installation check script https://github.com/jackhumbert/qmk_firmware/blob/master/util/Win_Check.bat. That should hopefully display an error, and if not the script should be improved.

Anyway running `mingw-get install msys` should fix it.

closed 2017-06-27 by skullydazed