qmk/qmk_firmware #787: Makefile support arguments to specify device

rpbaltazar, 2016-09-26:
Recently trying to flash my ergodox infinity, I ran into a problem with this step:

`sudo make infinity-keymapname-dfu-util` from the [instructions](https://github.com/jackhumbert/qmk_firmware/tree/master/keyboards/ergodox#ergodox-infinity)

as dfu-util would find multiple devices and didn't know which one to install to.
I've walked around the problem by installing it manually as described [here](https://github.com/rpbaltazar/qmk_firmware/tree/master/keyboards/ergodox#ergodox-infinity)
1. build
2. go to build folder
3. run the dfu-util with the params to specify the device.

It would be a nice to have a way to pass these arguments to the makefile.

fredizzimo, 2016-11-07:
I dont think its possible to do in a general way. We could possibly support a known subset of options, like specifying the device, but supporting arbitrary command line arguments is not possible, because make would try to parse them. So it would have to be implemented differently for each type of flashing command that we support.

Make doesn't even support variables containing strings with spaces from within makefiles, so thats not an option either.

drashna, 2018-10-21:
I believe that better handling was added to this, but I'm not sure.

However, for now, I'm closing this.  If it's still an issue, please reopen.

closed 2018-10-21 by drashna