qmk/qmk_firmware #791: Can we remove of CMP as a dependency in the toolchain?

IBNobody, 2016-09-29:
Every support comment I run into on Windows seems to revolve around `CMP` (part of diffutils) not being present.

In fact, it is the only hard line requirement that drives people to need MinGW as opposed to just the MHV AVR tools + dfu-programmer.

What does `CMP` do? It is used to generate some of the build breadcrumbs (*flags.txt / compiler.txt / obj.txt).

```
$1/cflags.txt: $1/force
	echo '$$($1_CFLAGS)' | cmp -s - $$@ || echo '$$($1_CFLAGS)' > $$@

$1/cppflags.txt: $1/force
	echo '$$($1_CPPFLAGS)' | cmp -s - $$@ || echo '$$($1_CPPFLAGS)' > $$@

$1/asflags.txt: $1/force
	echo '$$($1_ASFLAGS)' | cmp -s - $$@ || echo '$$($1_ASFLAGS)' > $$@

$1/compiler.txt: $1/force
	$$(CC) --version | cmp -s - $$@ || $$(CC) --version > $$@
endef

.PRECIOUS: $(MASTER_OUTPUT)/obj.txt
$(MASTER_OUTPUT)/obj.txt: $(MASTER_OUTPUT)/force
	echo '$(OBJ)' | cmp -s - $@ || echo '$(OBJ)' > $@

.PRECIOUS: $(MASTER_OUTPUT)/ldflags.txt
$(MASTER_OUTPUT)/ldflags.txt: $(MASTER_OUTPUT)/force
	echo '$(LDFLAGS)' | cmp -s - $@ || echo '$(LDFLAGS)' > $@
```

Is there a way to rewrite the statements to not use CMP?

IBNobody, 2016-09-29:
What most of those `CMP` statements are doing is seeing if the earlier version of the file matches the 
current data. If they do not match, it writes the current data to the file. 

Make is sensitive to file dates changing which makes regenerating the files automatically every time (via an ` echo '$$($1_CFLAGS)'  > $$@)` call a bad thing to do because it will trigger a full compile again even when none of the source files changed.

Likewise, you cannot simply comment out the file creation lines because Make will expect them to be there and not find them.

And for whatever reason, Make breaks altogether if you comment out any of those recipes.

jackhumbert, 2016-09-29:
I was going to bug @fredizzimo about this as well :) alternatively, if there's a way to document/suggest how to fix it more easily (before reaching the error), that would be acceptable as well.

IBNobody, 2016-09-29:
I don't know if documenting is the answer. The latest person who ran into this issue on /r/olkb was running multiple toolchains. If we say "use MinGW" and give instructions but they are using TDM-GCC, how would they know to fix this?

I guess telling them to install MinGW and MSYS would "fix" the issue, but that particular user would have a toolchain pulling tools from multiple folders! That would make things hard to troubleshoot if they ran into other problems.

fredizzimo, 2016-09-29:
The cmp command plays a very central role in the current make file structure. That was the reason why we were able start doing incremental building, and only build what's changed, and at the same time get rid of the quick make targets.

What it basically does is to check if some compilation flags have changed and if they have, it will trigger a rebuild. Otherwise it just rebuilds the changed files, of course taking header files includes into account. It's also able to detect things like compiler version changes, and force rebuilds when that happens as well.

So this allows us to detect the situation where someone modifies a variable in the makefile, and trigger a rebuild if needed. Using a pretty standard solution of adding all the input makefiles as dependencies would not work, since the user can also pass variables through the commandline

So unless we implement the flag comparison in another way, there's really no way to get rid of the cmp command and keep the same functionality.

What I suggest instead is that if we are on windows, we run win_check.bat as a part of the make process to report any errors in a nicer format. We could have similar checks for other operation systems as well. 

Another thing that should be ensured, is that the setup scripts works properly and install all that's required. At the same time it should perhaps be improved it so it doesn't need to use hacks to modify the path, and possibly corrupt your windows installation. I described one way of doing it in this [comment](https://github.com/jackhumbert/qmk_firmware/issues/526#issuecomment-233639137).

I won't have any time for doing these kinds of changes though, as I would like to allocate the extra time I have on making the effect system, so that the Infinity Ergodox becomes at least as usable as it was in my TMK based repository.

algernon, 2016-09-29:
I'm not familiar with what is included in the various windows tools, but... do we have anything in there that can compute the md5 or sha1 sum of a file? If so, the Makefiles could be changed to echo the stuff into a temporary file, and compare sums (either via the `[ ... ]` shell test thingy, or with `make`, jumping through a few hoops), instead of comparing content. If the sum differs, delete the old file, and rename the new. If its the same, keep it as-is.

This way the timestamp will only change when the contents changed, and we won't need `cmp`.

(I do not have a Windows system to test on, mind you. I'll happily implement the above change, if someone can tell me if there's a sum tool, and if I can bribe someone into testing it)

fredizzimo, 2016-09-29:
Yes, we could use the [`comp`](https://technet.microsoft.com/en-us/library/bb490883.aspx?f=255&MSPPError=-2147217396). But then the make file needs to detect if we are on windows, or at least fall back to that if `cmp` is not found, because the commands have different syntax.

Another thing that I forgot to mention, is that the problem might not only be the `cmp` command.  Both [this](https://github.com/jackhumbert/qmk_firmware/issues/782) and [this](https://github.com/jackhumbert/qmk_firmware/issues/709), also reports that date.exe crashes. So there might be another unrelated problem.

IBNobody, 2016-09-29:
Ugh... Date is a mess too. There's an MSYS date.exe command and Window's native date command. Win should always call native Date, even if MSYS has a date.exe.

Are there any other executables that I missed in Win_Check?

drashna, 2018-03-25:
Is this an issue anymore?

fredizzimo, 2018-04-07:
No, it isn't. Both msys2 and WSL correctly supports `cmp`

closed 2018-04-07 by fredizzimo