qmk/qmk_firmware #793: MAKE CLEAN needs cleaning up

IBNobody, 2016-09-29:
I have a few issues with MAKE CLEAN.

First, there is no simple `make clean` recipe anymore for the general QMK.

```
C:\GitHub\qmk_firmware>make clean
make: *** No rule to make target 'clean'. Stop.
Makefile:474: *** commands commence before first target.  Stop.
```

Second, I know that `make allkb-clean` works, but there's an issue in that it is sloooow because it iterates on every keyboard/keymap.

Third, there isn't a `make clean` command variant that gets rid of the hex files. I understand why we would want to keep hex files, but we don't always want them hanging around.

fredizzimo, 2016-09-29:
Yes, it has been on my long todo list, but I would be happy if someone else could do it,

You should never actually need to run clean though. If you have to, then there are some bugs in the dependency checks that needs to be fixed. And that should be reported as separate issues.

IBNobody, 2016-09-29:
I can look into tackling 1 and 3 this weekend.

You are right. You never truly need to run `make clean` for individual projects. However, it is the only way I know of to regenerate any compiler warnings.

For all projects, though, I usually want to clean up the .build folder because I don't need them taking up space.

drashna, 2018-03-25:
#2485 adds a "make clean" command. And simply deletes the entire .build folder, so its fast.

This handles `#1` and `#2`.  As for `#3`, `rm *.hex should fork for #3. 

I'm closing this.  If needed, reopen or open a new issue.

closed 2018-03-25 by drashna