qmk/qmk_firmware #795: Using other keys for `MT()`

iffsid, 2016-09-30:
What would be the simplest path to enable the use of some other keys (caps lock, etc.) as modifier keys?

I'm particularly interested in being able to use the `MT(mod, key)` macro with `KC_CAPSLOCK` for the special-purpose case of controlling my window manager. (I'm on Linux.)

This is a somewhat convoluted setup, but it appears to be required because the `MOD_HYPER` key as defined sends a combination of `ctrl+alt+shift+cmd` instead of triggering the `Hyper_L` keycode directly.
The `Hyper_L` availability is nice as I can assign it to the different `mod` keys that `X` makes available natively.

PureSpider, 2016-10-12:
I'd like to chime in here and request that both left and right modifier keys can be used with MT(), not only left ones.
My case requires right shift to be used as a modifier.

mrak, 2017-02-04:
It it a limitation of the bitshifting that prevents right modifiers from working?

PureSpider, 2017-02-14:
From what I see, 97816df7e7aa8710c8a0837b2abe008e0c765f2a adds this feature - at least the right side modifiers?

yanfali, 2017-09-18:
I'd like to chime in, my use case is I would like to be able to use MT(MO(4), KC_SPC) to implement a mouse layer on a split space bar keyboard. When the key is tapped it should act like a space, when it's held down it should allow a layer change. If someone can tell if this is possible with the current code, I may take a look at implementing it.

jcowgar, 2017-09-18:
@yanfali I believe what you are looking for is LT...

LT(layer, kc) - momentary switch to layer when held, and kc when tapped.

i.e., in your case, LT(4, KC_SPC)

yanfali, 2017-09-18:
Oh, thanks. Perfect.

closed 2018-10-21 by drashna