qmk/qmk_firmware #796: Repeated keypresses not registered in leader sequences

achin, 2016-10-01:
I'm trying to setup a leader sequence that has multiple, consecutive occurrences of the same key (e.g. "AA", "BB", "CC"), but I can't get it to work right. No matter what I tried, it seemed like repeated keys weren't be registered at all.

I finally started debugging the leader sequence array directly. I ran the following code and listened for results with hid_listen.

```
void matrix_scan_user(void) {
    LEADER_DICTIONARY() {
        leading = false;
        leader_end();

        print_hex16(leader_sequence[0]); print("\n");
        print_hex16(leader_sequence[1]); print("\n");
        print_hex16(leader_sequence[2]); print("\n");
        print_hex16(leader_sequence[3]); print("\n");
        print_hex16(leader_sequence[4]); print("\n");
    }
}
```

When I ran sequences without repeated keys, I got the output that I expected, but whenever I pressed the same key more than once, only the first key was registered. For example, when I pressed KC_LEAD, KC_A, KC_A, I got the following output.

```
0004
0000
0000
0000
0000
```

I'm using QMK version 5f9c2f6 on a hand-wired Minorca keyboard with a customized QMK keyboard.

drashna, 2018-03-25:
Is this still occurring?

drashna, 2018-10-21:
`SEND_STRING` would probably work here.

closed 2018-10-21 by drashna