qmk/qmk_firmware #801: Locking Caps Lock Not Working on macOS Sierra

djpyle, 2016-10-03:
I noticed the other day that locking caps lock was no longer working after upgrading to macOS Sierra. I was using TMK (on an ADB converter) at the time, and Hasu figured out a solution for me (there needs to be a slight delay between when caps locks registers and de-registers in action.c). Today, I noticed that QMK has the same issue, and the same solution fixes it. Here's a link to Hasu's original reply where he pinpointed the problem: https://geekhack.org/index.php?topic=14290.msg2268814#msg2268814

shelaf, 2016-10-03:
Ref. TMK issue
https://github.com/tmk/tmk_keyboard/issues/390

shelaf, 2016-10-03:
This issue is something to do with this feature on Sierra? http://apple.stackexchange.com/questions/255519/how-can-i-make-caps-lock-work-like-it-did-prior-to-sierra

tlvenn, 2016-10-07:
Hi @djpyle , are you able to compile the firmware still with Sierra and Xcode 8 ?

djpyle, 2016-10-07:
@tlvenn I haven't had any issues compiling.

icirellik, 2016-10-19:
@tlvenn I had issues today after running a `brew upgrade`.  Reinstalling `avr-libc` fixed the issue.

skullydazed, 2017-06-27:
I'm closing this to clean up our open issues, but if you still have questions @djpyle please reopen this or open a new issue. I can't find the commit right now but I believe this has been fixed.

closed 2017-06-27 by skullydazed