qmk/qmk_firmware #802: keyboards/ergodox/ez/twimaster.c:9:22: fatal error: inttypes.h: No such file or directory

alaibe, 2016-10-03:
The error appear after a full upgrade on the system:
osx ugraded to sierra and brew update and upgrade

do you need any other log?

tlvenn, 2016-10-03:
Same thing is happening for me with Sierra and Xcode 8

tlvenn, 2016-10-07:
@jackhumbert any insight on this particular issue ? Thanks

fredizzimo, 2016-10-08:
It's very hard to tell without knowing more details.

But my first guess is that it's somehow using the wrong compiler, or that the avr-libc isn't properly installed. The `twimaster.c` is the first file to be compiled, so that explains why it fails on that one.

I would try to  uninstall `avr-gcc` and `avr-libc` and then reinstall them. You might also get more support [here](https://github.com/osx-cross/homebrew-avr).

alaibe, 2016-10-10:
indeed just a remove/install makes it work

thx for the help

closed 2016-10-10 by alaibe