qmk/qmk_firmware #812: 'KC_0xF1' undeclared

treeman, 2016-10-09:
While trying to define a keymap for gh60 I get this error:

```
keyboards/gh60/gh60.h:38:17: error: 'KC_0xF1' undeclared here (not in a function)
     { KC_##K00, KC_##K01, KC_##K02, KC_##K03, KC_##K04, KC_##K05, KC_##K06, KC_##K07, KC_##K08, KC_##K09, KC_##K0A, KC_##K0B, KC_##K0C, KC_##K0D }, \
```

With the code:

```
    KEYMAP_ANSI(
        TRNS, F1,  TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS, \
        TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS, \
        TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,     TRNS, \
        TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,          TRNS, \
        TRNS,TRNS,TRNS,               TRNS,               TRNS,TRNS,TRNS,TRNS),
```

It seems that F1 gets expanded to 0xF1 as it's interpreted as a hex value which is taken literary in the macro expansion.

Using

```
avr-gcc (GCC) 4.9.2
```

Interestingly I can build the same code using the tmk firmware with the same compiler.

A workaround for the problem is to either use `KEYMAP(` directly or define a bunch of:

```
#define KC_0xF1 F1
```

fredizzimo, 2016-10-09:
I haven't tried it, but I think we need to force macro expansion with something like this

```c
#define CONCAT(a, b) a##b

#define KEYMAP(K00) \
{ \
    { CONCAT(KC_, K00),  } \
}
```

fredizzimo, 2016-10-09:
Sorry it needs to be done in two phases like this
```c
#define CONCAT(a, b) CONCAT2(a,b)
#define CONCAT2(a, b) a##b

#define KEYMAP(K00) \
{ \
    { CONCAT(KC_, K00),  } \
}
```

jackhumbert, 2016-11-16:
Sorry, this `KEYMAP` shortcut needs to be converted over to the standard structure (without the `KC_` prefix being added).

closed 2017-06-27 by skullydazed