qmk/qmk_firmware #816: data.exe stopped working

insertnamethere, 2016-10-10:
Ran the setup twice on a Windows 10 PC and a Windows 7 PC with the same result. Some error about data.exe stopped working. I can't even generate the Win_Check log cause i get the same issue

fredizzimo, 2016-10-13:
I guess the issue is the same as #782 and #709. And discussed in more detail in #791. 

I will try to see if I can add more checks to the make command in itself. I also think that the install script might need some fixup. But that will happen earliest this weekend.

fredizzimo, 2017-06-03:
I'm closing this, since there's already other similar issues reported. Also if you are using Windows 10, you can now follow [these instructions](https://docs.qmk.fm/Build-Environment-Setup.html#windows-10).

closed 2017-06-03 by fredizzimo