qmk/qmk_firmware #817: Two keys combination as a trigger for macro (or anything else)

pbogut, 2016-10-10:
Is it possible to map phisical key combination as a  trigger? I cant find information anywhere.

What I'm trying to achieve is to make ctrl + h send backspace. It would work in gui applications like it is working in terminal.
My idea was to set a key (my ctrl) as layer trigger, and then on new layer define every key as LCTR(KC_??) for everything except H which would be maped as backspace. But that sound like a very wird solution.
Is there any other way?

algernon, 2016-10-10:
The described solution would work, too, but there are a couple of other options. The easiest, I think, is to use a macro instead of `KC_H`: the macro would check if `Ctrl` is pressed, and if so, unregisters it, and presses backspace. Also saves the state control was in, so we can restore it on release. If `Ctrl` is not pressed, it registers `H`. On release, it would check whether control needs to be restored, and if so, unregisters `Backspace`, and re-registers `Ctrl`. If not, then it unregisters `H`.

It would look something like this:

```c
case M_H:
  static bool rereg_ctrl;
  if (record->event.pressed) {
    if (keyboard_report->mods & MOD_BIT(KC_LCTRL)) {
      rereg_ctrl = true;
      unregister_code (KC_LCTRL);
      register_code (KC_BSPC);
    } else {
      rereg_ctrl = false;
      register_code (KC_H);
    }
  } else {
    if (rereg_ctrl) {
      unregister_code (KC_BSPC);
      register_code (KC_LCTRL);
    } else {
      unregister_code (KC_H);
  }
  break;
```

Something like this. You can easily extend this to work for both controls: just use an `int` instead of a `bool`, and have `0` be the case where no control needs to be re-registered, `1` when you need the left one, `2` when you need the right. Or something along those lines.

Hope this helps.

algernon, 2016-10-10:
For reference, other solutions include playing with `process_record_user`, and catching `KC_H` there, along with `KC_LCTRL`/`KC_RCTRL`, and having some global state. It's pretty much the same logic as above, just elsewhere.

As far as I see, there is no functional difference between the two, but the macro version sounds easier to understand, and is a little bit shorter, perhaps.

DidierLoiseau, 2016-10-10:
Pay attention to when you release the control key before the H, as you could end up with a stuck control key. You probably need a macro for control as well to handle this.

pbogut, 2016-10-10:
@algernon thanks so much, I will play around with that. I'm running into the issue described by @DidierLoiseau so I need to sort it out. The tricky part is that my control is already working as ESC on tap.

algernon, 2016-10-11:
You could try a `wait_ms(50)` or so, between the register/unregister of control and `H`. Otherwise, if you use `process_record_user`, you can catch the `ESC/Control` press, and replace `H` there. Perhaps that'll help the situation, and is easier to catch there, than making a macro that both stores the control state, and does the `ESC/Control` dance.

pbogut, 2016-10-12:
I've ended up with two macros. One for `H` and one for `CTL/ESC`. I can't put everything in `process_record_user` as I still want to be able to press `ctrl+h` in rare occasions.

I will leave it here for reference. Thank you very much.
Macro
```c
        case M_CTL_ESC:
            if (record->event.pressed) {
                register_code(KC_LCTRL);
                ctrl_esc_timer = timer_read();
                ctrl_pressed = true;
                key_pressed = false;
            } else {
                unregister_code(KC_LCTRL);
                ctrl_pressed = false;
                if (timer_elapsed(ctrl_esc_timer) < 250 && !key_pressed) {
                    register_code(KC_ESC);
                    unregister_code(KC_ESC);
                }
            }
            break;
        case M_H:
            if (record->event.pressed) {
                if (ctrl_pressed) {
                    ctrl_h_bs_pressed = true;
                    unregister_code(KC_LCTRL);
                    register_code(KC_BSPC);
                } else {
                    ctrl_h_bs_pressed = false;
                    register_code(KC_H);
                }
            } else {
                if (ctrl_h_bs_pressed) {
                    unregister_code(KC_BSPC);
                } else {
                    unregister_code(KC_H);
                }
                if (ctrl_pressed) {
                    register_code(KC_LCTRL);
                }
            }
            break;
```
And in `process_record_user`

```c
    if (record->event.pressed) {
        key_pressed = true;
    }

```

closed 2016-10-12 by pbogut