qmk/qmk_firmware #820: LT() freezing keyboard on one layer

karptonite, 2016-10-12:
For this keymap, https://github.com/karptonite/qmk_firmware/blob/LT-bug/keyboards/planck/keymaps/karp/keymap.c , based on the default (relevant changes in line 57),
using the `LT(_RAISE, KC_RCBR)` key as a modifier (ie, not just tapping) freezes my planck. That is, if I tap the key, I get a `]` as expected. But if I hold it and type any other key, (a number, for example) they keyboard freezes and won't work again until I unplug and plug it back in. With the default firmware in place, the keyboard works fine.

Interestingly, the `LT(_LOWER, KC_LCBR)` does NOT freeze the keyboard. I'm new to keyboard firmware programming, so it is possible this is my mistake, but it seems worth reporting either way.

algernon, 2016-10-12:
I think the issue is that `KC_RCBR` and `KC_LCBR` use `Shift`, while `LT()`'s second argument is registered without modifiers. Using shifted symbols in the second position causes some funky behaviour.

karptonite, 2016-10-12:
how is `KC_RCBR` using shift? it is an unshifted character normally.

algernon, 2016-10-12:
My bad. I was going by the laptop keyboard in front of me, which has `{` as a shifted `[`. Looking at the QMK source, though, you are correct, it should not be a shifted char.

As such, I have no idea why it may lock up the keyboard.

jackhumbert, 2016-10-12:
@algernon is right, `KC_RCBR` and `KC_LCBR` (`{` and `}`) are shifted characters - you can see the definition [here](https://github.com/jackhumbert/qmk_firmware/blob/aa8aa6af75f7c7a6a66d83930f9bca306dfb8578/quantum/keymap.h#L241), which don't work with the `LT()` shortcut. `KC_LBRC` and `KC_RBRC` (`[` and `]`)are the lower case ones.

karptonite, 2016-10-12:
Ah, thanks, that makes sense!

closed 2016-10-12 by karptonite