qmk/qmk_firmware #821: Support Windows' Alt+DecimalANSICode for inputting special characters?

DidierLoiseau, 2016-10-12:
After struggling a lot trying to make UTF-8 work properly on Windows (if only I didn't have to use that for work…), I figured that the characters I needed are actually part of CP1252, and Windows has [a simple solution to enter those](http://www.georgehernandez.com/h/xComputers/CharacterSets/Shortcuts.asp#windows) (first method) which, as far as I can tell, is the most reliable one across applications:

> <kbd>ALT</kbd>+_DecimalANSICode_. Hold <kbd>ALT</kbd>, then type the ANSI value in decimal on the numeric keypad. If the number is preceded by a zero and is less than 256, Windows assumes you a decimal code point in the current keyboard/input language. For most Americans, this is a Windows ANSI code.
> 
> * EG: <kbd>ALT</kbd>+<kbd>0</kbd><kbd>1</kbd><kbd>6</kbd><kbd>5</kbd> yields a Yen symbol (¥) for its Windows ANSI character. This works in most apps.
> * EG: <kbd>ALT</kbd>+<kbd>0</kbd><kbd>1</kbd><kbd>3</kbd> yields a CR (Carriage Return) in Notepad but does nothing in Wordpad.

The current UTF-8 solution that we have for Windows in QMK is the 4th one, but AFAICT, it does not work in Office apps and RichEdit components. Basically it does not work in most Microsoft applications.

**Would it make sense to have a common function available in QMK so that anyone could easily reuse it, or should it remain part of each keymap?**

Here is my implementation (probably a bad naming as it actually works based on the user keyboard/input language, it could be something else than CP1252):
```c
void send_cp1252(uint16_t cp1252)
{
    // For more info on how this works, see first method of http://www.georgehernandez.com/h/xComputers/CharacterSets/Shortcuts.asp#windows
    // Pressing ALT followed by 0 and the decimal code point.
    register_code(KC_LALT);

    uint16_t remainder = cp1252;
    // voluntarily starting with i > 256 so that we start with a 0
    for (int i = 1000; i > 0; i /= 10) {
        int hex = remainder / i;
        remainder = remainder % i;
        register_code(hextokeycode(hex));
        unregister_code(hextokeycode(hex));
    }

    unregister_code(KC_LALT);
}
```

priyadi, 2016-10-24:
@DidierLoiseau you might want to take a look at the discussion at https://github.com/jackhumbert/qmk_firmware/issues/837

DidierLoiseau, 2016-10-24:
@priyadi the advantage of the solution I propose is that it does not need any additional software, and works in all applications (AFAICT).

The main inconvenient is of course the limited character set of CP1252 (or other code page one might be using).

I think it would make sense to support both UTF-8 and what I propose here, letting people choose what works best for them.

drashna, 2018-10-21:
The Unicode feature handles this. 

Also, it's more than just holding "Alt".  And the host may need some additional configuration to support this. 

https://docs.qmk.fm/#/feature_unicode?id=unicode-input-methods

closed 2018-10-21 by drashna