qmk/qmk_firmware #828: Bash having issues with the number of things to compile

jackhumbert, 2016-10-16:
This was first seen in #824, then #827, and now in master. It shows itself with a couple of different error messages, such as:

* `/bin/sh: 2: eval: Syntax error: Unterminated quoted string`
* `/bin/sh: 2: eval: Syntax error: end of file unexpected (expecting ")")`

Removing a certain number of keymaps/keyboards fixes the issue. All individual keyboards/keymaps still compile, so aside from not being able to test things with travis, this isn't a huge issue right now for people using QMK - just those trying to open PRs.

Dbroqua, 2016-10-16:
Hello,

After a quick test it's seems that the origin of the problem is chibios_test.

In order to debug I deleted keyboard/chibios_test and run the make all command (`TARGET=all-keyboards AUTOGEN=true && make $TARGET AUTOGEN=$AUTOGEN`) with success. 

Regards,
Damien

Dbroqua, 2016-10-16:
Ok,

By removing this keyboards : 

> chibios_test
> ergodox
> infinity60
> kitten_paw

Make exit with success.

I this directories one or many keymaps of this keyboards are in errors and will fail the make all.

jackhumbert, 2016-10-16:
You can remove any of the keyboards/keymaps and it will make - from what I can tell, it has to do with the number of them, not which ones are included.

Dbroqua, 2016-10-17:
Are you sure ?

Because if I remove all except the ones I have quoted before and I run `make` it fails too.

Regards

sethbc, 2016-10-18:
set VERBOSE=true, it's running an absolute monster of a command at Makefile:490.  This needs to be broken down to multiple commands.  I think this is what @fredizzimo was working on in terms of making the test on travis fail while still compiling everything at once.

pvinis, 2016-10-18:
could you paste the command so we see? i saw somewhere that we could use a
`foreach`, so if the monster command is just maaaany arguments, we can just
put all these in a for and then its done.  
  
\--

Pavlos Vinieratos

![](https://link.nylas.com/open/353mq70q5h26xxffslxc5ml3/local-48f17979-e591?r
=cmVwbHkrMDAwMTg3ODliYzhiMmVkMzYxNTBmZTg5YTgzOGIzNDI5MjkxMDQ0YzlhYjc0NDhlOTJjZ
jAwMDAwMDAxMTQxZTIxNzc5MmExNjljZTBhZWNjODljQHJlcGx5LmdpdGh1Yi5jb20=)

> set VERBOSE=true, it's running an absolute monster of a command at
Makefile:490. This needs to be broken down to multiple commands. I think this
is what [@fredizzimo](https://github.com/fredizzimo&r=cmVwbHkrMDAwMTg3ODliYzhi
MmVkMzYxNTBmZTg5YTgzOGIzNDI5MjkxMDQ0YzlhYjc0NDhlOTJjZjAwMDAwMDAxMTQxZTIxNzc5Mm
ExNjljZTBhZWNjODljQHJlcGx5LmdpdGh1Yi5jb20=) was working on in terms of making
the test on travis fail while still compiling everything at once.

rt/qmk_firmware/issues/828#issuecomment-254584158&r=cmVwbHkrMDAwMTg3ODliYzhiMm
VkMzYxNTBmZTg5YTgzOGIzNDI5MjkxMDQ0YzlhYjc0NDhlOTJjZjAwMDAwMDAxMTQxZTIxNzc5MmEx
NjljZTBhZWNjODljQHJlcGx5LmdpdGh1Yi5jb20=), or [mute the
kIU1UViqbnnBks5q1QV3gaJpZM4KYE7C&r=cmVwbHkrMDAwMTg3ODliYzhiMmVkMzYxNTBmZTg5YTg
zOGIzNDI5MjkxMDQ0YzlhYjc0NDhlOTJjZjAwMDAwMDAxMTQxZTIxNzc5MmExNjljZTBhZWNjODljQ
HJlcGx5LmdpdGh1Yi5jb20=).![](https://github.com/notifications/beacon/AAGHiec6Q
xVRze0CFwE4sGGUqtSV9aYVks5q1QV3gaJpZM4KYE7C.gif)

jackhumbert, 2016-10-18:
@sethbc thanks - I hadn't seen that. I'm seeing the call come from the `BUILD` function(?) on Makefile:386, though.

fredizzimo, 2016-10-18:
I didn't have time to look at it yesterday. But I will today after I have made some food and got something to eat.

It's not as simple as splitting it to multiple commands it also needs to detect the errors correctly, and support the makefile -j flags correctly.

jackhumbert, 2016-10-18:
I'm going to remove planck/rev3 temporarily, so we can continue to test/merge things. Thanks for chiming in @fredizzimo.

fredizzimo, 2016-10-18:
Just a small update, getting rid of the error is easy, for example by defining RUN_COMMAND like this 
```
# The empty line is important here, as it will force a new shell to be created for each command
# Otherwise the command line will become too long with a lot of keyboards and keymaps
define RUN_COMMAND 
$(COMMAND_$(SILENT_MODE)_$(COMMAND))

endef
```

However that means that each keymap build is done in it's own subshell. So the tricky part is to get the status back to the main makefile. Variable exports only works from parent to subshells so that's not an option. Any attempt to capture the exit code is also problematic, since that would also happen inside a subshell. 

The `shell` makefile function can not be used either as that captures the output. So even if the output could be printed later, it would mean that we don't get real-time output anymore. 

Any other ideas? 

The real problem is that the main makefile(just the `makefile` and not the ones that do the actual building) is trying to do something that make isn't designed to do. Make doesn't really have any concept of ordering and building a list of things, so it took a lot of hacks and awkward code to do what it did, which worked until we hit this limit. In a real programming language like `python` this would just take a few lines of code, and anyone would understand what it does, however that would mean bringing in yet another dependency. 

Converting it to a standard shell script might also be possible, but I don't have much experience in sh or bash programming to tell how hard that would be.

Finally there's other options like cmake, but I don't know if that supports this kind of things either.

Furthermore doing the conversion no matter which of the options are chosen, would probably mean that it the commands for building keyboards would yet again need to be changed, which doesn't seem like a good idea.

So let's hope that we find a solution inside make.

pvinis, 2016-10-18:
im trying to understand *why* this monster command is needed.. it has the same thing for all keyboards. so somehow the keyboard list is generated, then this monster command is generated, then its ran. why not generate the keyboard list, and then iterate it and make one by one? makes no sense to have all of them in one command.
i wanted to try that, but this makefile is so huge and so hard to follow.. ive seen way huger projects with much smaller and manageable makefiles.
we have to either replace this madness with something that makes the whole thing easier to run and understand, or write this makefile from the beginning, this time more modular.

jackhumbert, 2016-10-18:
@fredizzimo I like the idea of temporary files, as long as we can successfully read the statuses quickly.

fredizzimo, 2016-10-18:
I will have to give up for today. Googling around didn't turn up anything. I might have some time to continue before work tomorrow, otherwise it has to be a few hours in the evening again.

So, so far the best solution seems to be to write a temporary file with the status. In the long run, yet another makefile overhaul using either Python, a shell script or possibly cmake, but that's not something that I have time to do.

@pinvis, It's needed because we are abusing the makefile system to run sequential commands instead of letting it manage all the dependencies and run everything in whatever order it wants. Additionally in order to do be able to catch errors from one of the commands we need to run them all inside the same subshell, therefore generating a huge command.  

Another complication that we have, which make the makefile non-standard, is that we need to compile the same files multiple times, possibly using different compilers. Making a non-recursive makefile for something like that is a very hard task, and if we did, we would get the output in a random order.

BTW. I already showed a way to have it generate one command for each keymap, in my message above this. Simply adding a new line would make it do that, but as explained it wouldn't catch any errors.

pvinis, 2016-10-18:
aha, ok i kind of get it now. and why do we actually need all keyboard to be built in sequential order? why not let make run all the things however it likes? if anything fails, then make fails anyway. alphabetic order is not so important.

sethbc, 2016-10-18:
if i recall, we initially talked about this in #573.  he idea was that it would be better to have travis run an entire build and catch errors at the end.

fredizzimo, 2016-10-27:
Sorry for not having time to fix this yet. I have been on a business trip, and still will be for a few days. So I don't have access to tools for compiling and testing. This means that the fix probably won't happen until the next week.

PR #857 (2016-10-30): Run each sumbmake in a separate shell
This will fix the situation where the commandline gets too long when there are many keyboards and keymaps.

Fixes #828
closed 2016-10-30 by jackhumbert