qmk/qmk_firmware #835: Keyboard with large matrix randomly hangs

IBNobody, 2016-10-19:
My 6 row x 22 column keyboard running on an AT90 chip is having some odd issues in Windows 10. It randomly drops out and eats key up commands. In troubleshooting, I see that USB communication drops out. Even hidlisten fails, as you can see below.

```
keyboard_report: 00 00 08 00 00 00 00 00
keyboard_report: 00 00 00 00 00 00 00 00
keyboard_report: 00 00 0F 00 00 00 00 00
keyboard_report: 00 00 19 00 00 keyboard_report: 00 00 4F 00 00 00 00 00
keyboard_report: 00 00 00 00 00 00 00 00
keyboard_report: 00 00 19 00 00 00 00 00
keyboard_report: 00 00 00 00 00 00 00 00
keyboard_report: 00 00 0F 00 00 00 00 00
```

I caught the error happening with the Microsoft Message Analyzer (USB packet sniffer) and saw that the main endpoint was being re-enumerated. 

I have tried disabling everything I could via the Makefile, but no luck. Any tips on how to proceed?

IBNobody, 2016-10-19:
I am attempting to see if the system can be broken by repeatedly registering and unregistering keystrokes. I will also try the keyboard on a second computer with logging enabled.

fredizzimo, 2016-11-05:
@IBNobody, does the matrix scan changes in your pull request fix the issue?

IBNobody, 2016-11-07:
@fredizzimo I do not believe it does. While I have not run into the issue since I made my changes, I also didn't have all the devices connected to my PC that I did when I had the issue occur. I will attempt to replicate the environment this week and let you know.

I did not see a case where this issue would have been fixed by my code, however, one thing my code did address was to make the ROW2COL matrix logic be more memory efficient. Maybe I was running out of memory? (But then why would my keyboard continue to work after a pause?)

IBNobody, 2016-11-16:
I have not been able to narrow down the fault of this issue to the keyboard. Signs point to some instability in the entire USB hub structure of my PC at work. I will close this issue and reopen it when I start seeing more errors that I can attribute to the keyboard/firmware.

closed 2016-11-16 by IBNobody