qmk/qmk_firmware #840: Instructions on burn tada68 firmware

TheLevenCreations, 2016-10-22:
Hi,

I am a Tada68 user. I just found @TerryMathews commits a pull request as follows, which is for tada68:
https://github.com/jackhumbert/qmk_firmware/pull/778

May I ask, what's the instructions to burn this firmware? 

Where tada68 is delivered, there is a website http://123.57.250.164:3000/tada68  used to reflash the firmware. It can download a bin file, then use it to replace the FLASH.BIN in usb when tada68 is switched to program mode.

So, can I just do the following?

1) use keyboards/tada68 source code to generate a bin file
2) replace this bin file in usb
3) tap ESC in tada68 keyboard

Because that website can only change the key settings, it can't change led settings. So I want to burn a firmware myself. Would you like to give me some tips, please? Thanks.

PureSpider, 2016-10-24:
Yes, run `make flashbin` and it generates a flashable file for you that you can flash just like the one you downloaded from the website.

TheLevenCreations, 2016-10-24:
@PureSpider Great, thanks for your reply! Will have a try later:)

TerryMathews, 2016-10-25:
> tada68 used to reflash the firmware. It can download a bin file, then use

When you are ready to compile QMK, make flashbin will give you a FLASH.BIN
that you can drag over to your keyboard when it's in bootloader.

TheLevenCreations, 2016-11-15:
Thanks for your instructions! I use 'make flashbin' command to generate a FLASH.BIN and it works quite well! :)

Thank you so much! You guys are so great! :-)

TheLevenCreations, 2016-11-15:
@TerryMathews

Not sure if you know the following information about tada68. I just found them by searching github.com:

https://github.com/jichuntao/online_compiler
https://github.com/jichuntao/tmk_keyboard

Under https://github.com/jichuntao/tmk_keyboard there is a 'kb68' keyboard source file which points to be tada68. Probably jichuntao is the author of tada68?? However, the pin configuration in this source is different.

closed 2016-11-15 by TheLevenCreations

TerryMathews, 2016-11-15:
The code that was contributed was tested on a production R1 TADA68 from
Originative. It's correct.

PureSpider, 2016-11-15:
Can confirm, I am running my R1 TADA68 with QMK without any problems.

rundeploy, 2017-05-02:
I replaced the new generated .bin file by qmk and replaced on the keyboard. So what's next ?