qmk/qmk_firmware #845: Ignore Matrix Keys

IBNobody, 2016-10-24:
I ran into an issue on my build where I needed to ignore keys on my matrix. I have key switch suppressors that snap onto my keys and make them act as "dead" keys. Without this change, I was unable to put my computer to sleep.

Here is my solution, in a nutshell.

Keymap
```
#define MATRIX_MASKED
const matrix_row_t matrix_mask[MATRIX_ROWS] =
{
//  1098765432109876543210987654321
  0b0000000001111111101111011111111,
  0b0000000001111111111111111111111,
  0b0000000001111111111111111111111,
  0b0000000001111111111111111111111,
  0b0000000001010111111111111111111,
  0b0000000001111101111111101011111,
};
```

```
matrix_row_t matrix_get_row(uint8_t row)
{
    // Matrix mask lets you disable switches in the returned matrix data. For example, if you have a
    // switch blocker installed and the switch is always pressed.
#ifdef MATRIX_MASKED
    return matrix[row] & matrix_mask[row];
#else
    return matrix[row];
#endif
}
```

OferPlesser, 2016-11-25:
Wouldn't defining the suppressed keys as KC_NO in your keymap suffice?

IBNobody, 2016-11-25:
No. That only makes keys do nothing when pressed. It does not block the
detection of them being pressed.

Wilba6582, 2016-11-25:
Is this the cause of the issue you had with the backlight staying on while in suspend mode?

IBNobody, 2016-11-25:
This was a different issue. There is code in the suspend logic that scans
the matrix to see if a key is pressed. Any key fires it, even KC_NO. If key
blanks are used, they will be detected as key down events and cause the
host computer to be woken up.

piotr-dobrogost, 2016-11-25:
> I have key switch suppressors that snap onto my keys and make them act as "dead" keys.

How come you did not attach picture of them? :-)

IBNobody, 2016-11-26:
> Sorry, I realised that a few seconds after I posted (and edited my
> comment).
>
> FWIW something like this would avoid the need for the mask, if the keymap
> has KC_NO on the blocked keys:
> bool suspend_wakeup_condition(void) { matrix_power_up(); matrix_scan();
> matrix_power_down(); for (uint8_t r = 0; r < MATRIX_ROWS; r++) {
> matrix_row_t matrix_row = matrix_get_row(r); for (uint8_t c = 0; c <
> MATRIX_COLS; c++) { if (matrix_row & ((matrix_row_t)1<<c)) { if
> (keymap_key_to_keycode(0, (keypos_t){ .row = r, .col = c }) != KC_NO) {
> return true; } } } } return false; }

There were other ways to handle it, but having a matrix mask was the least
intrusive. It has to be enabled, and it is very process efficient.

IBNobody, 2016-11-28:
@piotr-dobrogost 

Here:

http://pimpmykeyboard.com/switch-blocker-pack-of-10/

They snap onto the switch base. They always keep the switch pressed down.

piotr-dobrogost, 2016-11-28:
Thanks for picture. What's the purpose of disabling some keys? Btw, you described these as *dead* keys which has specific meaning in the context of keyboards – https://en.wikipedia.org/wiki/Dead_key

IBNobody, 2016-11-28:
I have a grid keyboard, and I wanted to mask off some keys. For example, I have a row of 14 keys that I use for my F keys. I have these switch blockers installed between F4 and F5, and between F8 and F9. They make the keyboard easier to use because I can find the right F key blindly. 

It also allowed me to convert those blocked switches into normal switches if I ever wanted to use them. I just pop off the blockers and put on a keycap.

But... Law of unintended consequences... I didn't realize that QMK would sense them as pressed and wake the computer. By masking the matrix, I can force them to be inactive.

IBNobody, 2016-11-28:
I am closing this issue, though because I implemented the mask.

closed 2016-11-28 by IBNobody