qmk/qmk_firmware #850: Space Cadet Shift not working on German Layout and LT()

tobsn, 2016-10-26:
I've tried everything to make it work and it just doesn't for the right shift.
Left shift seems to work just fine, right shift can't be mapped that way.

I tried a work around with:
LT( LAYER_RSFT, LSFT( DE_OSX_LESS ) ),

But the layer cancels out the left shift and it's always just DE_OSX_LESS (#define DE_OSX_LESS KC_GRV)

jgouly, 2016-10-27:
LAYER_RSFT ends up as 2, shifted by 8 in the LT macro. Which is the same as QK_LSFT  (1<< 8)!

tobsn, 2016-10-27:
Explanation:

```
#define LT(layer, kc) (kc | QK_LAYER_TAP | ((layer & 0xF) << 8))

kc = DE_OSX_MORE = LSFT(DE_OSX_LESS) = 0x0200 | DE_OSX_LESS = 0x200 | 0x35 = 0x235
layer = 2
layer_shifted = 0x2 << 8 = 0x200
QK_LAYER_TAP = 0x4000

0x235 | 0x4000 | 0x200 = 0x4235
```

closed 2017-06-27 by skullydazed