qmk/qmk_firmware #853: Debouncing is being triggered on multiple keystrokes

IBNobody, 2016-10-28:
I do not think that the debouncing routine is implemented correctly. Or at least it is not being triggered correctly on a large keyboard with 22 columns.

I ran into the following case...

```
keyboard_report: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
bounce!: 02
bounce!: 01
bounce!: 03
```

... when I was typing. The debounce ate all my keystrokes.

Pressing keys individually does not trigger a debounce event, but pressing multiple keys in combination or in rapid/overlapping succession will cause a debounce event to erroneously trigger.

IBNobody, 2016-10-28:
As a follow-on, it seems that setting the debounce interval does not work as intended.

```
/* Debounce reduces chatter (unintended double-presses) - set 0 if debouncing is not needed */
#define DEBOUNCING_DELAY 1
```

If you set the delay to 0, the following code breaks...

```
    for (uint8_t i = 0; i < MATRIX_ROWS; i++) {
        select_row(i);
        wait_us(30);  // without this wait read unstable value.
        matrix_row_t cols = read_cols();
        if (matrix_debouncing[i] != cols) {
            matrix_debouncing[i] = cols;
            if (debouncing) {
                debug("bounce!: "); debug_hex(debouncing); debug("\n");
            }
            debouncing = DEBOUNCING_DELAY;
        }
        // unselect_row(i);
        unselect_rows();
    }

    if (debouncing) {
        if (--debouncing) {
            wait_ms(1);
        } else {
            for (uint8_t i = 0; i < MATRIX_ROWS; i++) {
                matrix[i] = matrix_debouncing[i];
            }
        }
    }

```

...because the debouncing loop never is entered to set the matrix!

IBNobody, 2016-10-29:
I am in the process of rewriting matrix.c. I have optimized the reads for row2col. I have separated out the scanning from the debouncing.

IBNobody, 2016-11-01:
@ezuk Can you check out my PR regarding the debounce / matrix.c rewrite? I tested it using some of Jack's boards, but I wasn't able to test it with yours. I do not want the PR to be merged until I can get it tested on multiple configurations because of how critical the matrix code is.

ezuk, 2016-11-03:
@IBNobody thank you -- I looked at it, but this is really in @jackhumbert's wheelhouse. My general guideline here is "omg please don't break my keyboard that thousands of people rely on". :) 

I am happy to compile and test; but I don't understand the code at all. A wise man once said "never 👍 PRs you are utterly clueless about" (and if not, that should really be a Chinese proverb). So I'm gonna wait on Jack to peruse through the Matrix changes in depth and maybe even test with an EZ (if you're up for it, Jack).

closed 2016-11-19 by IBNobody

piotr-dobrogost, 2016-11-24:
@IBNobody

What's the status?