qmk/qmk_firmware #855: Infinity60 v1.1 with ChibiOS scanning slowly

tianminc, 2016-10-30:
I flashed my Infinity60 v1.1 (with LEDs) with QMK firmware and found it responses slowly and drops keys when I type fast. I investigated a bit and found each call to matrix_scan() in keyboard.c takes 9ms. That does not sound very right. I did not modify anything (except for adding print in keyboard.c to debug it). Any suggestions? Thank you in advance!

Depariel, 2016-11-02:
In `config.h`, try changing `#define DEBOUNCE    5` to `#define DEBOUNCE    0` and see if that helps at all.

tianminc, 2016-11-02:
Thanks for the reply. I tried changing DEBOUNCE to 0, but it does not make a difference. AFAIK the firmware works well with Infinity60 1.0. I am very curious about why merely changing matrix_scan() will make it slow down so much.

tianminc, 2016-11-02:
Update: I tried it on my Windows 10 machine and it works without a problem even in rubbing key test. The machine that I had problem previous was running Ubuntu 14 with 3.13.0-100 kernel. That's interesting.

closed 2016-11-03 by tianminc

fredizzimo, 2016-11-05:
@tianminc, you are correct the matrix scan is very slow on the Infinity60. I have added a pull request in #866, which should fix the issue. The problem is that I don't have the keyboard to test, so it would be great if you could test the changes for me.

tianminc, 2016-11-06:
@fredizzimo Thanks for the update. Though in my case, the main cause of slow scan seems to be related to console output. After I disabled debug console, the keyboard becomes much more responsive and totally usable. I will try it later and report back.