qmk/qmk_firmware #858: Build Concurrency

skullydazed, 2016-10-30:
I spent a little time this weekend to work on the [build system](https://github.com/skullydazed/qmk_configurator/commit/e036ee217d25b7b13c2122d2400cc94abf4a855b) for my configurator app. I now need to figure out how to deploy this in production. My primary concern right now revolves around concurrency- what happens if two builds are running at the same time in the same source tree?

Some context-

* When a user triggers a compile, the RQ job creates a new keymap file, `keyboards/<keyboard_name>/keymaps/<hashid>/keymap.c`.
* The RQ job then chdir's into `qmk_firmware` and runs `make KEYBOARD=<keyboard_name> SUBPROJECT=<subproj> KEYMAP=<hashid>`

This works great when I trigger a single build. Objects are compiled, my firmware is linked, and everything is happy.

What if I have two keymap compiles kicked off at the same time? Will they try to write object files to the same location? Do I need to take care to ensure that only one compile is happening at once?

fredizzimo, 2016-10-30:
Running two build in parallel would unfortunately not work. 

First of all there's a shared version.h file, that is generated each time you run make.

But ignoring that, before pull request #857 it would have been possible to run two different keyboards in parallel. Even two keymaps of the same keyboard would have worked for all keyboards except the ChibiOS ones, which share the ChibiOS  related object files in order to speed up the compilation. 

Pull request #857 broke that since there's now a global shared file for indicating errors. 

For single keymaps you can archive parallelism by specifying the -j# command line option to make. And I guess it would also be possible to build multiple keyboards in parallel, with some small changes to the makefiles, so that you would be able to specify where the version.h file is generated, and being able to have a different output directory for each time you run it.

skullydazed, 2016-10-30:
Thanks. I'll approach this from the standpoint of having only 1 build per tree. Maybe this is a good excuse to look into building a Docker image that can pull build jobs from my queue. :)

closed 2016-10-30 by skullydazed