qmk/qmk_firmware #859: Tap Dance + OSM

kthibodeaux, 2016-11-01:
When I try to use tap dancing with OSM it doesn't work.  If I have `ACTION_TAP_DANCE_DOUBLE(KC_Z, OSM(MOD_LSFT))` it does nothing.  If I have `ACTION_TAP_DANCE_DOUBLE(KC_Z, OSM(MOD_LGUI))` then the second tap outputs `e`.

I also tried setting it up in the action like i have SCLN and SPC but got nothing there, either.

The end goal is to have zz make the next keypress modified by CMD, but tapping then holding z act as CMD until released.

My full setup is here: https://github.com/kthibodeaux/ergodox-layout/blob/tap-dance-z-for-osm-cmd/keymap.c

algernon, 2016-11-01:
`ACTION_TAP_DANCE_DOUBLE` expects both arguments to be plain keycodes, with modifiers applied at best. It does not support other actions like OSM. To have an OSM effect applied, you'll have to use your own custom tap-dance functions.

Something along these lines:

```c
case TD(Z):
  if (state->count == 1) {
    register_code (KC_Z);
    unregister_code(KC_Z);
  } else {
    set_oneshot_mods (MOD_LGUI);
  }
  break;
```

kthibodeaux, 2016-11-01:
Thanks algernon, you rock!

closed 2016-11-01 by kthibodeaux