qmk/qmk_firmware #863: Implications from switching unicode input mode at runtime?

PureSpider, 2016-11-04:
I recently developed a feature to be able to change the unicode input mode in my keymap without needing to recompile the firmware and so far it seems to work great.

I was however wondering if there are any side effects to changing the input mode on runtime as opposed to only setting it once when the firmware is loaded? Maybe someone that is more familiar with QMK can give some insights on this.

The code in question is here: https://github.com/PureSpider/qmk_firmware/blob/master/keyboards/tada68/keymaps/purespider/keymap.c#L87-L169

algernon, 2016-11-04:
Nope, there is no difference between changing the unicode input mode at keyboard boot time, at compile time, or in response to a keypress. All it does under the hood is switch an integer to another value, nothing else.

PureSpider, 2016-11-04:
That's what I thought, just wondered if I overlooked something because the readme explicitely says "this is currently limited to supporting one OS at a time, and requires a recompile for switching." [here](https://github.com/jackhumbert/qmk_firmware/blob/master/readme.md#unicode-support)

fredizzimo, 2016-11-07:
I think that part of the documentation is outdated

At least it was written [way before](https://github.com/jackhumbert/qmk_firmware/commit/04a36ecbf400959d6c320a120ea5a092847dd9b6) the [current implementation](https://github.com/jackhumbert/qmk_firmware/commit/b732b79b49b098dba8e14493c745075f336747d8), which was further refactored by @algernon a bit later

closed 2017-06-27 by skullydazed