qmk/qmk_firmware #864: S60-x vendor information doesn't display properly as a usb device on Linux

Eleuin, 2016-11-05:
i expect that both the device ID and its vendor information would show up but only the device ID shows up

`lsusb` output:
```
Bus 001 Device 002: ID 8087:8001 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 043: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 002 Device 023: ID 8087:07dc Intel Corp. 
Bus 002 Device 004: ID 064e:920b Suyin Corp. 
Bus 002 Device 061: ID feed:6060  
Bus 002 Device 002: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

`dmesg | tail` output:
```
[57916.027398] usb 2-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[57916.027399] usb 2-3: Product: S60-X
[57916.027400] usb 2-3: Manufacturer: Massdrop
[57916.029101] input: Massdrop S60-X as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/0003:FEED:6060.005E/input/input108
[57916.081170] hid-generic 0003:FEED:6060.005E: input,hidraw1: USB HID v1.11 Keyboard [Massdrop S60-X] on usb-0000:00:14.0-3/input0
[57916.082110] input: Massdrop S60-X as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.1/0003:FEED:6060.005F/input/input109
[57916.082343] hid-generic 0003:FEED:6060.005F: input,hidraw2: USB HID v1.11 Mouse [Massdrop S60-X] on usb-0000:00:14.0-3/input1
[57916.083247] input: Massdrop S60-X as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.2/0003:FEED:6060.0060/input/input110
[57916.134829] hid-generic 0003:FEED:6060.0060: input,hidraw3: USB HID v1.11 Device [Massdrop S60-X] on usb-0000:00:14.0-3/input2
[57916.135541] hid-generic 0003:FEED:6060.0061: hiddev0,hidraw4: USB HID v1.11 Device [Massdrop S60-X] on usb-0000:00:14.0-3/input3
```

Eleuin, 2016-11-05:
even more curious is the verbose usb out
```
Bus 002 Device 064: ID feed:6060  
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0xfeed 
  idProduct          0x6060 
  bcdDevice            0.01
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
```

fredizzimo, 2016-11-07:
The problem is that it costs $5000 to get a valid vendor id. http://www.oshwa.org/2013/11/19/new-faq-on-usb-vendor-id-and-product-id/

That said we could [apply](http://wiki.openmoko.org/wiki/USB_Product_IDs) for product ids. Before that we should probably make sure that we conform to the GPL license, which we have to use because of the TMK roots and library use.

skullydazed, 2017-10-22:
I'm going to close this issue for now. We're well versed in what the issues surrounding this are, and for the time being it doesn't seem like QMK as an organization has a viable path forward to resolve this. If that should change in the future we'll revisit this issue.

closed 2017-10-22 by skullydazed