qmk/qmk_firmware #875: Tap dance doesn't work on ergodox

jackvnimble, 2016-11-12:
I've been trying to get a simple double tap working and can't seem to get it to work.
I removed the line from the `quantum/process_keycode/process_tap_dance.h` that checks to see if it's enabled because for whatever reason I can't get the setting to carry over from the Makefile

(it's literally just typing `TAP_DANCE_ENABLE=yes` into the Makefile right?.. anyways)

I got my keymap.c to compile and flashed my keyboard and the key I tried to enable double-tap on doesn't work at all. Here is the file: https://gist.github.com/jackvnimble/6ab9447e74a1f448dfce3c126888cc9c
 
Relevant code is around line 20 and line 50.

algernon, 2016-11-12:
Does it input `4` on single tap at least? As in, does it not work at all, or does it input the wrong stuff? My gut feeling is that `LCTL(LGUI(LSFT(KC_4)))` is a bit too much for it, but before suggesting a workaround, I'd like to see what is wrong exactly.

jackvnimble, 2016-11-12:
Nope, single tap doesn't do anything either.

algernon, 2016-11-12:
Interesting. In this case, I'd think that you did not properly remove the ifdef guards... Can we try getting the Makefile stuff work?

Having a Makefile in the same dir as your keyboard, with `TAP_DANCE_ENABLE=yes` ought to do the trick. If that does not work either, if you could show me your fork of the firmware would be helpful.

krasnovu, 2016-12-06:
@algernon, sorry for my english. I have the same problem on ergodox.
If i enable tap dance in my Makefile
`TAP_DANCE_ENABLE=yes`
and define in keymap.c
```
enum tap_dance {
  TDCTRL = 0,
};...[skipped]...

..., TD(TDCTRL), ...

qk_tap_dance_action_t tap_dance_actions[] = {
  [TDCTRL]  = ACTION_TAP_DANCE_DOUBLE (KC_Q, KC_W)
};
```
that compiling fine, but if i flash hex file to keyboard
they initialize by fading leds but then not working any keys in defined base keymap

pattulus, 2017-01-14:
After having "Make finished with errors" and failing with the examples provided in the Wiki I came across this open issue.

I copy and pasted the code except for replacing it with a more basic test: `[TD_TEST] = ACTION_TAP_DANCE_DOUBLE(KC_0, KC_1)`. What can I say, it works on my EZ. I didn't have to go to `quantum/process_keycode/process_tap_dance.h` either – just added the default oneliner to my Makefile.

drashna, 2018-03-25:
jack, you still having this issue? 

If so, try without the modes, just the normal keycodes (as I know that a number of keys have issues with mods... eg don't support them, and that may be the case here).

jackvnimble, 2018-05-16:
@drashna, sorry for the delay in response here. I'm not actively trying to get this to work any longer and have no intention of trying to reflash my EZ as I'm having trouble with my infinity now and need at least one keyboard with my custom map 😆 

I'll assume the key combo I was attempting was not compatible as patrickwelker seems to have gotten it to work with something a bit simpler.

Feel free to close. Again, sorry for the delay

drashna, 2018-05-16:
Sounds like you need more keyboards. :P 

And yeah, I'll close it for now.

closed 2018-05-16 by drashna