qmk/qmk_firmware #889: MT(mod, kc) send kc instead of mod when the key has tapped -> hold

lambdalisue, 2016-11-19:
I've using a `MT` macro to make a key which send `Ctrl` when hold, and `Space` when tapped. And the definition said

> MT(mod, kc) - is mod (modifier key - MOD_LCTL, MOD_LSFT) when held, and kc when tapped. In other words, you can have a key that sends Esc (or the letter O or whatever) when you tap it, but works as a Control key or a Shift key when you hold it down.

However, I noticed that when I tap a key and then hold, `kc` is continuously sent instead of `mod` even the key has **hold**.
This is so annoying for example (underscores `_` are used instead of spaces):

1. I've typed `They_are_a`
2. **Tap** `Ctrl/Space` to enter a space, the text become `They_are_a_`
3. **Hold** `Ctrl/Space` and `H` (`Ctrl-H`) to remove letters before the cursor but the text become `They_are_a__hhhhhhh` instead of `They_are_`

Note that `TAPPING_TERM` which might be related to the behavior above didn't solve the situation.
I've tried `100`, `200` (Default), and `300` but the problem was persistent...

Is this a bug or am I missing some configurations? Any helps would welcome.

Thanks.

---

I am using ErgoDox EX and my configuration is on
https://github.com/lambdalisue/qmk_firmware/tree/master/keyboards/ergodox/ez/keymaps/pinkyless

NoahAndrews, 2016-11-21:
To me, this is desirable. I use MT for my space key, and this provides a way to hold down the space key. Maybe there should be a way to configure this or something?

I'm not the only one who likes it the way it is: https://www.reddit.com/r/olkb/comments/5ckn3v/quick_question/

lambdalisue, 2016-11-21:
> To me, this is desirable. I use MT for my space key, and this provides a way to hold down the space key. Maybe there should be a way to configure this or something?

I see. Yeah I wish if the behavior could be controlled by some configuration.

lambdalisue, 2016-11-30:
In case, I solved this issue by modifying `tmk_core/common/action_tapping.c` as

```diff
diff --git a/tmk_core/common/action_tapping.c b/tmk_core/common/action_tapping.c
index e16e11b..bd9a69a 100644
--- a/tmk_core/common/action_tapping.c
+++ b/tmk_core/common/action_tapping.c
@@ -228,6 +228,7 @@ bool process_tapping(keyrecord_t *keyp)
         if (WITHIN_TAPPING_TERM(event)) {
             if (event.pressed) {
                 if (IS_TAPPING_KEY(event.key)) {
+#ifndef TAPPING_FORCE_HOLD
                     if (!tapping_key.tap.interrupted && tapping_key.tap.count > 0) {
                         // sequential tap.
                         keyp->tap = tapping_key.tap;
@@ -237,11 +238,11 @@ bool process_tapping(keyrecord_t *keyp)
                         tapping_key = *keyp;
                         debug_tapping_key();
                         return true;
-                    } else {
-                        // FIX: start new tap again
-                        tapping_key = *keyp;
-                        return true;
                     }
+#endif
+                    // FIX: start new tap again
+                    tapping_key = *keyp;
+                    return true;
                 } else if (is_tap_key(event.key)) {
                     // Sequential tap can be interfered with other tap key.
                     debug("Tapping: Start with interfering other tap.\n");
```

Then add the following into your `config.h`

```c
#define TAPPING_FORCE_HOLD
```

closed 2016-11-30 by lambdalisue