qmk/qmk_firmware #901: avr/pgmspace.h: No such file or directory

wyudong, 2016-11-24:
I was able to build my keymap.hex file several months ago on my Mac. But today when I try to build my new keymap.c, I get the following errors when running `make keymap=my_keymap`:

```
avr-gcc (GCC) 6.2.0
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  22852	      0	  22852	   5944	s60-x_donkey.hex

Compiling: ../../keyboards/s60-x/s60-x.c                                                           In file included from ../../keyboards/s60-x/s60-x.h:23:0,
                 from ../../keyboards/s60-x/s60-x.c:1:
../../quantum/quantum.h:5:26: fatal error: avr/pgmspace.h: No such file or directory
 #include <avr/pgmspace.h>
                          ^
compilation terminated.
 [ERRORS]
 | 
 | 
 | 
make: *** [../../.build/obj_s60-x_donkey/keyboards/s60-x/s60-x.o] Error 1
```

I tried to reinstall avr libs using brew but nothing changes. Can anyone help me to resolve it? Thanks.

wyudong, 2016-11-25:
If I run `make` in any keyboards/<keyboard> directory, the same error always occurs.

NoahAndrews, 2016-11-25:
Maybe try running `avr-gcc -E -x c++ - -v </dev/null` (see [this link](http://forum.sonic-potions.com/discussion/306/building-firmware-on-osx/p1)) and see if something seems amiss with one of the paths it outputs.

wyudong, 2016-11-25:
Reinstalling avr-libc solves my problem.

```
brew untap osx-cross/avr
brew uninstall avr-libc
```

Then 
```
brew tap osx-cross/avr
brew install avr-libc
```

Thanks @NoahAndrews. Issue can be closed.

NoahAndrews, 2016-11-25:
You should be able to close it yourself (I can't) :)

closed 2016-11-25 by wyudong