qmk/qmk_firmware #909: Stuck state for shifted punctuation

jasone, 2016-11-26:
I'm experimenting with an Ergodox-EZ [layout](http://configure.ergodox-ez.com/keyboard_layouts/qbdaoq) for which all standard keyboard punctuation is available without directly using a shift key.  As I train myself to use this I sometimes forget to not hold the shift key, and this causes the keyboard to enter a completely unusable state for !$% (shifted 1, 4, and 5).  Other shifted punctuation keys (e.g. @) cause less obvious layer switching issues that also require a keyboard reset.

Is there some alternate encoding for shifted punctuation that can avoid these issues, or is there a way to fix this in general?

jasone, 2016-11-26:
This appears to be related to using a _right shift_ key in combination with the shifted punctuation.  With the left shift key the problem does not occur.

skullydazed, 2017-06-27:
I'm closing this to clean up our open issues, but if you still have questions @jasone please reopen this or open a new issue.

closed 2017-06-27 by skullydazed

thanatos, 2017-08-28:
I also have this issue; I'm currently experiencing it with a customized ErgoDox EZ's firmware that I got from using their layout configurator though, so I'm not sure if it is (and I suspect it isn't) the latest version of QMK. (I need to figure out how to compile QMK…)