qmk/qmk_firmware #920: Output too large after recent changes

NoahAndrews, 2016-11-29:
I was wanting to play around with the just-merged SysEx API, but when I try to flash the updated firmware with [my custom keymap](https://github.com/NoahAndrews/qmk_firmware/tree/master/keyboards/planck/keymaps/noah), I get an error saying "Bootloader and code overlap", implying that the firmware is now too large. I suspect this is due to #918.  Perhaps there should be a way to exclude the new API from the firmware to save space.

jackhumbert, 2016-11-29:
I think it's setup to disable it via `API_SYSEX_ENABLE=no` in the Makefile. I'm also looking to minimise the footprint though :)

jackhumbert, 2016-11-29:
Oops - the one in rules.mk should have a `?=` - that should allow you to override it, I believe (it might not be possible otherwise)

NoahAndrews, 2016-11-29:
It worked fine as-is actually, thanks

closed 2016-11-29 by NoahAndrews

cbbrowne, 2016-12-01:
Can we have it disabled by default, and only enable it if it's set to "yes"?  That would mean we don't run into this a bunch of times when various people update their favorite layouts.