qmk/qmk_firmware #922: Something has grown?

cbbrowne, 2016-12-01:
I just had a scary moment; did a "git pull jack", "make dfu" on my favorite keymap (cbbrowne), and got the following...

Making planck/rev4 with keymap cbbrowne and target dfu

avr-gcc (GCC) 4.9.2
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text    data     bss     dec     hex filename
      0   29916       0   29916    74dc planck_rev4_cbbrowne.hex

Compiling: keyboards/planck/keymaps/cbbrowne/keymap.c                                               [OK]
Compiling: ./tmk_core/common/command.c                                                              [OK]
Linking: .build/planck_rev4_cbbrowne.elf                                                            [OK]
Creating load file for Flash: .build/planck_rev4_cbbrowne.hex                                       [OK]

Size after:
   text    data     bss     dec     hex filename
      0   29916       0   29916    74dc planck_rev4_cbbrowne.hex

dfu-programmer: no device present.
Error: Bootloader not found. Trying again in 5s.
Bootloader Version: 0x00 (0)
Bootloader and code overlap.
Use --suppress-bootloader-mem to ignore
tmk_core/avr.mk:110: recipe for target 'dfu' failed

I stepped back to the last time I messed with my code, and it built/installed fine, thus:

# git checkout cc7604d11670cf3155d53f0019557b249b1192c0
HEAD is now at cc7604d... Add breathing, as initial implementation with the RAISE/LOWER keys.  And throw in extra macros to shorten keymaps
root@cbbrowne:/home/cbbrowne/GitStuff/qmk_firmware/keyboards/planck/keymaps/cbbrowne# make dfu
Making planck/rev4 with keymap cbbrowne and target dfu
(note: did the *actual* build as cbbrowne, so all this is doing is the DFU run)

avr-gcc (GCC) 4.9.2
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text    data     bss     dec     hex filename
      0   24450       0   24450    5f82 planck_rev4_cbbrowne.hex

Compiling: keyboards/planck/keymaps/cbbrowne/keymap.c                                               [OK]
Compiling: ./tmk_core/common/command.c                                                              [OK]
Linking: .build/planck_rev4_cbbrowne.elf                                                            [OK]
Creating load file for Flash: .build/planck_rev4_cbbrowne.hex                                       [OK]

Size after:
   text    data     bss     dec     hex filename
      0   24450       0   24450    5f82 planck_rev4_cbbrowne.hex

Bootloader Version: 0x00 (0)
Validating...
24450 bytes used (85.27%)

I speculate that 29K of code is too much, where 24.5K is fine.

I know I have MIDI turned off, and took recommendations (seen on Reddit threads) of adding
#define NO_DEBUG
#define NO_PRINT

Nevertheless, I speculate that some of the substantial revisions to the qmk/tmk stuff over the last couple months has made something "bloat."

Further, I suggest that maybe there should be some check within the build to make sure that the load file isn't too large.

jackhumbert, 2016-12-01:
This is likely due to #918 - if you add `API_SYSEX_ENABLE = no` to your `Makefile`, things should drop back down.

cbbrowne, 2016-12-01:
That works, but I'm still inclined to leave this open as an issue, as:

1. Better to have API_SYSEX_ENABLE turned off by default, turn it on only explicitly
2. It would be nice to have some size measurement to guard against oversized binaries earlier in the process

jackhumbert, 2016-12-01:
I'm aiming for it to be enabled by default for all boards - right now it's only turned on for the Planck and the EZ. I believe the only reason it's causing trouble is because your Makefile (and a lot of others) is overriding some of the defaults that cut down on the size.

You can see the hex size when compiling (`74dc`, needs to be < `7000`), which is an easy way to tell how big things are before trying to flash.

I know you commented on it, but this is really a dup of #920 as well.

cbbrowne, 2016-12-01:
Yeah, it's mostly duplicative of #920, closing.

closed 2016-12-01 by cbbrowne