qmk/qmk_firmware #928: jeebak planck keymap generates too-large .hex, needs build parm changes

cbbrowne, 2016-12-02:
Creating load file for Flash: .build/planck_rev4_jeebak.hex                                         [OK]

Size after:
   text    data     bss     dec     hex filename
      0   31810       0   31810    7c42 planck_rev4_jeebak.hex

7c42h > 7000h, so that probably won't load. Suggestion: Take a look at parms in Makefile, it probably needs to have the Makefile revised to deactivate API_SYSEX_ENABLE, or similar.

jackhumbert, 2016-12-02:
I appreciate you finding these, but we really don't need an issue for each one of them - if you wanna list them all in one issue (or your PR even), that would be awesome.

cbbrowne, 2016-12-02:
They might be apropos to push upstream to the original author, that's reason to separate issues.  But I'm happy to take your lead on it.

jackhumbert, 2016-12-02:
I don't think that's possible through Github, and this is the upstream. We've generally had the policy that it's up to the contributor of the keymap to keep their code up-to-date, except where structural things are modified (this change may or may not qualify as that). Most of these keymaps were added in PRs like yours, and large portion of them aren't super active on here. Some of them are for hand-wired Plancks, which have different memory capacities as well.

jackhumbert, 2016-12-02:
Addressed in #929

closed 2016-12-02 by jackhumbert