qmk/qmk_firmware #937: Document options for mousekeys

NoahAndrews, 2016-12-06:
I never knew the mouse keys functionality was configurable. We should add documentation for that.

brandenbyers, 2016-12-06:
Configurable? As in change default speed and such like that? Or do you mean something different? I would definitely like to see some documentation on this. Where does this configuration take place?

brandenbyers, 2016-12-06:
Are these options what you are referring to? https://github.com/jackhumbert/qmk_firmware/blob/a258358b6c2df0d1347a6506831076ef6a9784b3/tmk_core/common/mousekey.c#L36-L53

NoahAndrews, 2016-12-06:
Yep. [See this commit](https://github.com/jackhumbert/qmk_firmware/pull/936/commits/a05b6a72d9da9fa6056046c9f600957f43b3c8df) for an example of their use.

bennypowers, 2017-03-12:
@NoahAndrews so modifying config.h as such and recompiling would provide for example, a faster mouse key experience?

```c
 #ifndef KEYBOARDS_ERGODOX_CONFIG_H_
 #define KEYBOARDS_ERGODOX_CONFIG_H_
  
 #define MOUSEKEY_INTERVAL       20
 #define MOUSEKEY_DELAY          0
 #define MOUSEKEY_TIME_TO_MAX    5
 #define MOUSEKEY_MAX_SPEED      5
 #define MOUSEKEY_WHEEL_DELAY 0
  
  #define TAPPING_TOGGLE  1
```

NoahAndrews, 2017-03-14:
Yep, that should work. 

I recommend creating a config.h inside your keymap folder, so it only affects the one keymap. Put these lines at the top of the file:
```
ifndef CONFIG_USER_H
#define CONFIG_USER_H
#endif

#include "../../config.h"
```

bennypowers, 2017-03-14:
You missed the leading `#`

NoahAndrews, 2017-03-14:
Sorry, fixed.

commit b2222053 (2017-07-03): Rewrite mouse_keys to resolve #937
closed 2017-07-07 by jackhumbert