qmk/qmk_firmware #938: Bluetooth with Planck and Handheld Scientific keyboard adapter module

MicahElliott, 2016-12-07:
Hi! I hope this is the right place to ask about this...

I got my hands on one of these really cool gadgets, the Handheld Scientific BT KB adapter: http://handheldsci.com/kb

I tried it out with some cheapo USB keyboards and it paired/worked great when prompted for a number code plus Enter.  However, I can't get my Planck to pair with it.  I don't know if it's the number sequence or the Enter that's not registering, or if anything is.  And I'm not sure how to debug it.  My question for now is simply: do you know of any reason such a key entry wouldn't work with the Planck?  Does it need to draw more power than this little module can give, or is there some incompatibility I've created with my LT bindings to Enter?  I can try going back to the default Plack layout with a normal Enter if you think that might help.

(I'm aware of this, but just not ready/able to go this route yet: https://www.reddit.com/r/MechanicalKeyboards/comments/3psx0q/the_planck_keyboard_with_bluetooth_guide_and/ )

jackhumbert, 2016-12-07:
I think I remember seeing one with a Planck before, but I wasn't able to find it on r/mk. You might need to disable the mouse keys in order for it to work - it mentions something about mice/keyboard combos not working, but I'm not sure it it's actually looking for those endpoints. 

It also says `max current 200mA`, and [the default is 500](https://github.com/jackhumbert/qmk_firmware/blob/master/tmk_core/protocol/lufa/descriptor.c#L297) - that might be something to try to change.

MicahElliott, 2016-12-07:
Thanks for the ideas.  I first tried removing the mouse keys, and that showed some promise.  Bluetooth showed it had accepted my number code, but said it was wrong.  Subsequent attempts didn't register at all, though.

So then I tried lowering the current and `make`:

```
...
Compiling: ./tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/Events.c                         [OK]
Compiling: ./tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/HostStandardReq.c                [OK]
Compiling: ./tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/USBTask.c                        [OK]
Linking: .build/planck_rev4_default.elf                                                             [OK]
Creating load file for Flash: .build/planck_rev4_default.hex                                        [OK]

Size after:
   text	   data	    bss	    dec	    hex	filename
      0	  30740	      0	  30740	   7814	planck_rev4_default.hex

dfu-programmer: no device present.
Error: Bootloader not found. Trying again in 5s.
Bootloader Version: 0x00 (0)
Erasing flash...  Success
Checking memory from 0x0 to 0x6FFF...  Empty.
Bootloader and code overlap.
Use --suppress-bootloader-mem to ignore
make: *** [tmk_core/avr.mk:112: dfu] Error 4
```

It managed to erase the flash, but then failed. Now the keyboard is not showing any signs of life (no beep, no light), so I may have bricked it.  Is there any way to hard reset it at this point to get the current back to 500?

jackhumbert, 2016-12-07:
It looks like the file's just too big, but unfortunately it only tests the size after erasing the flash memory. You should be able to flash a layout still if you try a different file/disable options to make smaller. The hex size should be under 0x7000 bytes.

NoahAndrews, 2016-12-07:
The error you got ("Bootloader and code overlap") means that your hex file was too big to flash. Your keyboard is fine, just currently firmwareless.

MicahElliott, 2016-12-08:
Doh. It's fine now. Turned off the audio in the Makefile to save a lot of space.

I'm still unable to get the bluetooth to work at 200mA and mouse keys disabled. It's funny that the very first time I disable mouse keys, it showed signs of accepting a code.

Thanks for the help, anyway.

MicahElliott, 2016-12-09:
Just wanted to report that with another Planck I have, this just worked (on the android I was messing with before).  I got the number-pair code right the first time and it was accepted and worked very nicely.  Wow, it's so nice to not have that ridiculous virtual keyboard covering the screen.

Anyway, I'll report here sometime if I figure out what's different between the two Plancks.  Great news that this thing works!

kalupa, 2017-04-05:
This same set up used to work for me on an older version of the firmware. Unfortunately I don't know "when" I built that version (nor do I have the hex file) but it was the stock version on the controller purchased in the Massdrop August 2015 GB.

I don't have backlights on the board but I think the `default` layout does, now.

Guess I'll have to try fiddling with C again but at least I'm not the only one trying this setup.

I'll also add my iPad refuses to see it when using a usb->lightning connector, too, so I'm guessing the mouse keys are not playing well based on this Issue. Will update if I have any luck

kalupa, 2017-04-17:
For any follow ups: lowering power requirements from `500` to `100` as per [this thread](https://www.reddit.com/r/olkb/comments/4i7moh/ipadiphone_power_consumption/) worked to resolve Handheld Sci BT and lightning adapter problems on this rev3 board.

closed 2017-06-27 by skullydazed