qmk/qmk_firmware #939: Print messages on a hex file being too large for the device

jackhumbert, 2016-12-07:
I'd like to automate the file size comparison a bit by having the max size of a hex file being stored as a variable in rules.mk, and if the current size is above it, it'll print an error/warning, and avoid continuing to the flashing (and therefore erasing w/o flashing) the chip.

I'm opting for a variable rather than a constant 0x7000 max for AVR chips, because the Teensy has a different max, and we do support other flash sizes.

I'm confident on how to implement this on the AVR chips, but not with anything else - if the max size isn't defined, the comparison will just be skipped.

cbbrowne, 2016-12-07:
That sounds appropriate, and yes, I certainly like the variable maximum, that being friendly to non-planck devices.

drashna, 2018-03-25:
Well, this is half done.  AVR boards are taken care of.  Just need ARM support.

drashna, 2024-10-12:
This can probably be closed now.  AVR has this, and works well.   And for ARM, while it's not a great solution.... the ld scripts generally error out if the compiled firmware exceeds the set flash size.  For instance: 
> | /Users/drashna/.local/qmk/toolchains/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: .build/zsa_voyager_drashna.elf section `.rodata' will not fit in region `flash0'
> | /Users/drashna/.local/qmk/toolchains/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: region `flash0' overflowed by 33328 bytes

closed 2024-10-12 by drashna

sigprof, 2024-10-13:
The problem on ARM is that if you are using a part of the internal flash to store the EEPROM data (which is the default for most chips), the LD script does not take the EEPROM backing store into account, therefore you can compile a firmware which overwrites a part of its own code when it starts up.  About the only exception are the STM32F4xx chips, where the flash layout is not really compatible with the default QMK behavior of placing the EEPROM backing store at the end of flash, therefore some space in the first 4×16KiB flash sectors gets reserved instead.

Also for most chips the flash size is not specified explicitly, and therefore some “default” size is assumed, which sometimes does not match the real hardware.  One especially problematic case is STM32F303xC vs STM32F303xB, because in this case the smaller chip also has less RAM, so the code works either by chance (because the nonexistent RAM is not actually accessed), or because the extra RAM actually exists, like the extra flash on STM32F103x8 (but is not guaranteed to work properly).