qmk/qmk_firmware #940: How to write switch-case for 16-bit keycodes (shift/alt/ctrl/layer combo-tap keys don't work)

MicahElliott, 2016-12-07:
I'm trying to set up a mapping where tapping Right-Shift produces a colon, like so:

    #define SFT_COLN SFT_T(KC_COLN)  // Tap for Colon, hold for Shift (does not work!)

But it doesn't work.  Instead, I get a semicolon (unshifted ;: key) when tapping. The same happens for other "LT" combos, like trying to map LPRN wrongly types a 9 (unshifted "9(" key).  Both of these are the unshifted version of the actual key.  So it seems we need a way to get the shifted key.  There are a few other keys I'd like to get this working for, like `: ( ) _ * -`.

I don't mind having to hack up the `process_record_user` switch statement if necessary. Some instructions on making this work would be really helpful.

As a workaround, I presently use some ugly xmodmap commands to get the keys I need, but it's not a universal solution.

I got some feedback from Jack on this directly:

> the SFT_T function only works with 8 bit keycodes, and the KC_COLN
> one is a 16 bit one, since it also applies the shift key, before
> sending the KC_SCLN key. The KC_LPRN works the same way, and
> unfortunately isn't compatible with most of those functions. Messing
> with process_record_user is unfortunately the only way to get that
> functionality right now, but it does give you a bit more control
> over what's happening :) there's an example in the readme.

----

So this below is a follow-up, with my present attempt at mapping "left-shift" as itself when held and "colon" when tapped.  Following the [readme](https://github.com/jackhumbert/qmk_firmware/wiki#bool-process_record_uint16_t-keycode-keyrecord_t-record), I'm this far:

    bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    switch (keycode) {
    ...
    case KC_LSFT:
      print("seeing left-shift");
      if (record->event.pressed) {
        register_code(KC_COLN);
      } else {
        // if (record->event.time > 100) {  // FIXME: IS THIS KINDA RIGHT??
        unregister_code(KC_COLN);
        // register_code(KC_LSFT); }  // THIS TOO??
      }
      return false;
      break;

This works for creating a `:`, but I need help with the timer for getting LSFT to register.  Is that FIXME line headed in the right direction?

Hard-coding the 100ms feels hacky (I could live with it), but I'm not sure where to get the default ms delay time.

I haven't tried exactly the code above, but various trials on the theme got me into cases of infinitely repeating `:`s and other oddities, so I need help at this point.

Once I have this case working, I'll plan to add more `case`s for the other other `KC`s I'm interested in.

I have `sudo hid_listen` working to see the print statements, but I'm unable to put any data into the strings (sprintf etc don't seem to work with the macro), so it's difficult to print-debug.

The description above only covers `SHIFT`, but I take it the same approach should affect/work with `ALT`, `CTRL`, and even layer multi-purpose keys.

JosephFerano, 2017-03-16:
👍 
Really interested in this as well, as I would like to map semi colon to colon, send semi colon when shifting, and when holding, switch to layer 3. Unfortuantely, `LT(3, KC_COLN)` this doesn't work. Any help would be appreciated.

drashna, 2018-03-25:
Yeah, I believe that the macros for this are not able to handle shifted keycodes. :(

Adding a solution for this would yeah, involve a timer. 
https://docs.qmk.fm/faq_keymap.html#timer-functionality

But you'd want to set the timer in the pressed action, but nothing else (remove the `register_code`). use the `matrix_scan_user` to check the timer.   Then for the "else" in the keypress, check the mod state (get_mod(), IIRC)... then clear the mod, and register and unregister the keycode, if needed. 

That should get this working.

drashna, 2018-10-21:
To clarify here, to add this sort of support natively, we'd need to expand the size of the keycodes for all of QMK.  Which isn't a small task. 

I'm closing for now, as several solutions have been posted here.

closed 2018-10-21 by drashna