qmk/qmk_firmware #941: Allow interrupting LT like MT

bendk, 2016-12-09:
For both layer tap and modifier tap there's a case where it's ambiguous what the user wants:

- user presses LT/MT key
- user presses other key
- user releases LT/MT key
- user releases other key

In this situation, does the user want to apply the layer/modifier to the second key, or do they want to have two separate keypresses?  There's no one correct solution.

What's frustrating is that the code is much different for LT and MT

- For LT this always registers as two key clicks
- For MT this usually registers as a modified click
  - Except if IGNORE_MOD_TAP_INTERRUPT is defined in which case it registers as two key clicks

I can see a couple solutions:

- Add a ENABLE_LAYER_TAP_INTERRUPT define, which works as the reverse of IGNORE_MOD_TAP_INTERRUPT.  This is the ugliest solution, but it doesn't change the default behavior
- Change the default behavior of LT to match MT and add IGNORE_LAYER_TAP_INTERRUPT, or make IGNORE_MOD_TAP_INTERRUPT apply to both.  This seems pretty reasonable to me, but it changes the default behavior. 
- Make it possible to set the behavior for each LT/MT keycode individually.  This would be the best IMO, but I don't know if there's a bit to spare.

drashna, 2018-03-25:
Is this still an issue?

bendk, 2018-03-27:
I can't comment on that, I stopped using the LT code and rolled by own instead.

drashna, 2018-03-27:
Okay. 

Would you mind posting a link to your code for others to reference? 
If not, that's fine! Figured I'd ask, though.

bendk, 2018-03-27:
I'm a little bit too ashamed to post it, especially since lately it seems to be causing double keystrokes for some reason.  I have a feeling that the LT behavior might have just gotten fixed along the way anyways.

drashna, 2018-03-28:
You should see some of my code.  "graceful" isn't exactly the word for it.  

merging in a keymap would allow somebody else to see it, and maybe improve upon it.

bendk, 2018-03-28:
So, I was all ready to share my code, then I actually don't even have the interrupt behavior enabled anymore.  Sorry, but I really have nothing to share.

drashna, 2018-03-28:
if you had it on github and committed, it would be in the history ...

drashna, 2018-10-21:
Okay.  Not a problem.

And I'm going to close this, for now.

closed 2018-10-21 by drashna

francislan, 2019-07-12:
I have the same issue and seems like the behavior hasn't changed from the description. The first two proposed solutions seem reasonable to me. I'll look into the code and see if I can implement them.

bendk, 2019-07-15:
For what it's worth, I finally pushed my code.  Here's how I implement LT:
https://github.com/qmk/qmk_firmware/blob/master/keyboards/ergodox_ez/keymaps/bdk/keymap.c#L91

francislan, 2019-07-15:
@bendk Thanks for your code! I was looking into action_tapping.c and action.c but I will just go with a modified version of your code to implement the behavior I want. I'm still too unfamiliar with the codebase (especially since there's almost no docs or comments) to make something more generic.

francislan, 2019-07-17:
For reference here's my custom implementation of LT that allows interruption. This is far from perfect (I listed a known "issue" in the comments) but this solves my issue at least.
https://github.com/francislan/qmk_firmware/blob/lan_keymap/keyboards/thevankeyboards/minivan/keymaps/lan/keymap.c#L237