qmk/qmk_firmware #950: Layer status leds

mrlinuxfish, 2016-12-14:
Upon upgrading the firmware on my planck keyboard to qmk from tmk so I can make use of nkro, I have lost the ability to control leds I have wired as layer status leds. I used to have a few lines in /tmk_core/common/led.h to allow this functionality. 
`
void led_layer_set(uint32_t state);
`
Then I would call my led.c file upon compilation which would read the layer state and light up various leds on the keyboard. 
```
// Led layer toggle
void led_layer_set(uint32_t state) {
    DDRB |= (1<<7); //led 0 (B7)
    //dprint("Set LEDs for layer change\n");

    // Led for Layer 1
    if ((1<<1 & state) != 0) {
        //dprint("Layer 2: on\n");
        PORTB |= (1<<7);
    } else {
        //dprint("Layer 2: off\n");
        PORTB &= ~(1<<7);
    }
	
    DDRB |= (1<<3); //led 1 (B3)
    //dprint("Set LEDs for layer change\n");

    // Led for Layer 4
    if ((1<<4 & state) != 0) {
        //dprint("Layer 2: on\n");
        PORTB |= (1<<3);
    } else {
        //dprint("Layer 2: off\n");
        PORTB &= ~(1<<3);
    }
	
    DDRB |= (1<<2); //led 2 (B2)
    //dprint("Set LEDs for layer change\n");

    // Led for Layer 2
    if ((1<<2 & state) != 0) {
        //dprint("Layer 2: on\n");
        PORTB |= (1<<2);
    } else {
        //dprint("Layer 2: off\n");
        PORTB &= ~(1<<2);
    }
	
    DDRB |= (1<<1); //led 3 (B1)
    //dprint("Set LEDs for layer change\n");

    // Led for Layer 3
    if ((1<<3 & state) != 0) {
        //dprint("Layer 2: on\n");
        PORTB |= (1<<1);
    } else {
        //dprint("Layer 2: off\n");
        PORTB &= ~(1<<1);
    }
}

```

The tmk project now uses a hooks file to streamline various calls, but this seems to have been added after the fork of this project.

Is it possible to implement this functionality again?

fredizzimo, 2016-12-27:
The easiest way to do that would be to implement  `void matrix_scan_user(void)` in your keymap, and put the above code there. Use the global variable `layer_state` to read the current layer. 

You might also want to implement `void matrix_init_user(void)` to init the ports.

Nephiel, 2017-05-09:
Just tried porting my custom board from TMK to QMK and came across this issue as well.
IMHO, setting layer leds on every matrix scan is overkill if they never change unless the layer changes.  
I solved it in tmk_core/common/action_tapping.c, by checking if the action is a layer change, and calling set_led() from there. I'm posting here so I don't forget to make a pull request later.

jpetermans, 2017-05-10:
I agree setting leds every matrix scan is too frequent, but you can also just check if layer_state has changed before setting leds. If no change, just return without running anything.

Implementing the tmk style hooks would be great though. I raised the issue (#1205) a little while back, and it sounds like there is potential to add them in future. I scanned through the related commit from tmk, and it is a little beyond me to pull together.

Nephiel, 2017-05-10:
I agree, hooks would be neat. But for now this approach allows to keep all LED-related code in the same places - `led_set_kb` and `led_set_user`. I just opened a pull request.

jackhumbert, 2017-06-27:
Since #1288 was merged, I think we can mark this as resolved - @mrlinuxfish, please reopen this if I'm mistaken!

closed 2017-06-27 by jackhumbert