qmk/qmk_firmware #954: USB Power draw advertisement

askreet, 2016-12-14:
I'm unable to plug my ErgoDox (using QMK) into an iPad Mini because it claims the power draw is too high. I can't imagine it's exceeding 500mA, so I'm wondering if someone could point me in the direction of fixing this in the firmware, or if I am way off base with how to approach this. I'm willing to learn and write the patch to implement this change with some help, if it's appropriate.

Thanks in advance!

Wilba6582, 2016-12-14:
Try changing:

`.MaxPowerConsumption    = USB_CONFIG_POWER_MA(500)`

in `tmk_core/protocol/lufa/descriptor.c` to use 100 instead of 500. If that doesn't work, try 50.

askreet, 2016-12-14:
Thanks for the tip -- I'll give that a shot.

Assuming it works, is this something qmk_firmware would want to record on a per-keyboard basis and bake into the resulting firmware? Is there a meaningful patch I could write to contribute this back to the community? Is "tmk_core" a vendored project I should be contributing to instead?

NoahAndrews, 2016-12-14:
TMK and QMK have pretty thoroughly split ways at this point. Code you contribute to tmk_core probably won't end up back here in a timely fashion.  I wouldn't be surprised if the max power consumption is already configurable by keyboard, but I'm not sure.

jackhumbert, 2016-12-14:
I'm not sure where we stand on making the standard 100 (it was a one point to accommodate for using boards with iOS devices). It would be nice to be able to configure it per keyboard via the `config.h` file.

Wilba6582, 2016-12-14:
I see no reason it shouldn't be configurable via `config.h` like the other USB descriptor info (i.e. name, desc, VID, PID) with a default of 500 if it isn't already defined before `descriptor.c`. i.e.

```#ifndef USB_MAX_POWER_CONSUMPTION
#define USB_MAX_POWER_CONSUMPTION USB_CONFIG_POWER_MA(500)
#endif```

askreet, 2016-12-15:
Thanks @Wilba6582, I've got a patch up in #955 -- I opted to put the call to the CONFIG_POWER_MA macro in the descriptor body so the default/override defines are in mA, let me know if you prefer your original suggestion to that and I can swap it out.

Wilba6582, 2016-12-15:
@askreet  Your way is better ;-)

closed 2016-12-18 by jackhumbert