qmk/qmk_firmware #959: Supporting ps2avrgb / Winkeyless.kr boards

ascii-soup, 2016-12-16:
It would be nice if I could run QMK on all my boards, and after asking on Reddit it seems like someone here might be able to make this possible.

Winkeyless.kr (B.face, B.mini etc) and by extension the 'clones' (often referred to as ps2avrgb after the firmware) PCBs use an ATmega32A4 from what I can see on my own board.

Can anyone here help out either adding support for these or maybe helping me to figure out how to do it myself? I haven't touched any microcontroller programming in a few years and even then it wasn't more than a dabble, but hey, who knows what's possible.

If there's any other information that's needed I'll do what I can to provide it.

Thanks,

jackhumbert, 2016-12-16:
For those digging deeper, it looks like [this is the schematic for the ps2avrGB](https://github.com/showjean/ps2avrU/blob/master/firmware/ps2avrGB/Circuit/ps2avrGB/20140829_140848_001.png) (found in [the repo](https://github.com/showjean/ps2avrU)) - this states it's an Atmega323, but I'm not sure if that's just an inaccurate label.

Thor77, 2016-12-18:
Really interested in this, too. If there's anything we can help with or if you can give any tips on how to port this firmware to a new device, please tell us :)

luizribeiro, 2016-12-22:
I did this for TMK. [Here's my code](https://github.com/luizribeiro/tmk_keyboard/tree/ps2avrGB/keyboard/ps2avrGB). I currently use my B.mini X2 with this firmware.

I tried getting it to work with QMK but I wasn't able to. I haven't had enough time to debug it though. I can probably put up my code so others can help out.

Maybe there's something on the V-USB protocol implementation that we need to patch so that it works again with QMK (I'm not aware of anyone who currently uses QMK with V-USB).

Lavosu, 2017-01-21:
I'm also interested in QMK for my B.Mini board. I look forward to progress on this!

luizribeiro, 2017-01-21:
Hey, I was able to get my B.mini X2 to run the QMK firmware today :)

The trick in the end was to use a custom matrix. The code is in [my fork of the qmk_firmware repo](https://github.com/luizribeiro/qmk_firmware/tree/ps2avrGB/keyboards/ps2avrGB). I'll get a pull request up soon, but first I want to polish the code a little bit and add some primitive backlight support.

I also had to make a few changes to `tmk_core`, so I'd be happy to hear feedback on those while I prepare a pull request. Most of the changes are so that the AVR core and the V-USB protocol support the ATMega32A, which is used by ps2avrGB boards. I also added a feature to the AVR core that allows us to reboot the board into the bootloader from software on the USB host, to make it more convenient to reprogram the board with a single command from the computer.

closed 2017-01-22 by jackhumbert

Lavosu, 2017-01-23:
Once I receive my board from my builder, I’ll start fiddling around with what you’ve provided. I’m glad someone has been able to do this! The option for Bootmapper or TMK/QMK is great.

jasonm23, 2017-02-12:
@luizribeiro did you manage to get the RGB under lighting working with QMK?

luizribeiro, 2017-02-12:
@jasonm23, I did! But I haven't made a patch that cleanly applies on master yet, it's currently only on my own branch of QMK which has my own keymap, etc. You can check out my current code for backlight here on this commit: luizribeiro/qmk_firmware@9dd1a3fe9b18165802a1b272faba362c8b54d9ad

I haven't put up a pull request to the main QMK repo yet mostly because I don't know what would be useful for people. On my personal branch I have the backlight change depending on the top-most active layer (I have a color setting per layer). Do you think that would be useful? I could either make a pull request for that or just enable backlight with a hard-coded color, and then people could customize the behavior on their own keyboards. What do you think?

jasonm23, 2017-02-12:
That's great.

Can we clarify, I'm interested in the RBG underglow rather than the per-key LED (which is what I think backlight is.)

Let me know what you've managed to get working.  (If I'm guessing correctly you're talking about per-key LED)

luizribeiro, 2017-02-12:
RGB underglow, I don't have per-key LEDs on my keyboard.

jasonm23, 2017-02-12:
Fantastic.

I have a b.pad and b.87 I will try the b.pad first.  Where did you get the schematic for the b.mini BTW?

luizribeiro, 2017-02-13:
I just read the source code of the original ps2avrGB firmware here: https://github.com/showjean/ps2avrU

While most of it is in Korean it wasn't too hard to figure out how the pins of the microcontroller are being used. That said, I believe all these boards use exactly the same pins, so I don't think you need to change anything on the matrix scan code to get it running.

All you have to do, AFAIK, is to create a custom `KEYMAP` macro for your board on `ps2avrGB.h` and create your own keymap.

Let me know how it goes! :)

digisax, 2017-06-05:
Do you know what I'd need to do to flash it onto a B.Mini EX? I have bootload and everything installed I'm just not sure if I need to change anything in the keymap files.

krusli, 2018-01-12:
I've previously ported over the underglow code from Luiz Ribeiro's personal fork of PS2AVRGB a few months ago, and recently just got backlighting working on the board (I'll still need to implement the "breathing" effect, but it should be really simple as the brightness control code is working).

It's in my [fork of the repo](https://github.com/krusli/qmk_firmware) in the `jj40` directory. The code (pinouts, settings) seem to be shared across all PS2AVRGB boards though. I'm going to open up a PR once I clean some things up as I had to make some changes to `quantum.c` to allow for custom backlighting functions (similar to `RGBLIGHT_CUSTOM_DRIVER`).

kaansoral, 2018-03-01:
Discussing FaceW here: https://geekhack.org/index.php?topic=58945.250 - Merlin64 mentioned the PCB heating with the custom QMK firmware, I wonder if anyone has similar experiences / opinions

PR #1021 (2017-01-22): Support for ps2avrGB / winkeyless.kr boards
This adds support for ps2avrGB / winkeyless.kr boards, using the V-USB protocol of the TMK firmware.

I had to make a few changes to `tmk_core` in order to get this working. To name a few:

1. Small changes to the Makefiles so that the V-USB protocol works again.
2. Added support for building with the `NO_UART` and `NO_SUSPEND_POWER_DOWN` options, since it made QMK a bit easier to build for the atmega32a these boards use.
3. Small changes to `tmk_core/common/avr/timer.c`. The atmega32a uses different register for some of the timers, so that file now  [... cut]