qmk/qmk_firmware #965: Macro not working right, possibly due to no time delay?

Anakey1, 2016-12-21:
https://github.com/Anakey1/qmk_firmware/blob/master/keyboards/planck/planck%20keymap.c

I have read before that there needs to be a delay before unregistering the shift to make custom shifted characters work properly so can someone please show me the extra code i would need to get the keymap to function correctly?

Thanks

fredizzimo, 2016-12-27:
Could you explain more what you are trying to do, and what exactly goes wrong? I couldn't really make sense of your keymap.

But for a delay, you can just add ` wait_ms(1)`, which would wait for 1 ms. I don't think it would be needed though. But if it is and 1 ms is not enough, you can also try with bigger timeouts.

Anakey1, 2016-12-28:
ok so i have set the SPECSHIFT layer to default as later 1.

i have the [ on the qwerty layer and i want to get ] on shift

I also want " on the default layer and ' on shift

however when i press shift and [ i still am just getting [ as if i had not pressed shift in the first place, simmilarly for the " key

fredizzimo, 2016-12-28:
Ah I made something similar for my keymap, which I haven't published yet. 

```c
void press_key_with_level_mods(uint16_t key) {
    const uint8_t interesting_mods = MOD_BIT(KC_LSHIFT) | MOD_BIT(KC_RSHIFT) | MOD_BIT(KC_RALT);

    // Save the state
    const uint8_t real_mods = get_mods();
    const uint8_t weak_mods = get_weak_mods();
    const uint8_t macro_mods = get_macro_mods();

    uint8_t target_mods = (key >> 8) & (QK_MODS_MAX >> 8);
    // The 5th bit indicates that it's a right hand mod,
    // which needs some fixup
    if (target_mods & 0x10) {
        target_mods &= 0xF;
        target_mods <<= 4;
    }

    // Clear the mods that we are potentially going to modify,
    del_mods(interesting_mods);
    del_weak_mods(interesting_mods);
    del_macro_mods(interesting_mods);

    // Enable the mods that we need
    add_mods(target_mods & interesting_mods);

    // Press and release the key
    register_code(key & 0xFF);
    unregister_code(key & 0xFF);

    // Restore the previous state
    set_mods(real_mods);
    set_weak_mods(weak_mods);
    set_macro_mods(macro_mods);
    send_keyboard_report();
}

void override_key(keyrecord_t* record, uint16_t normal, uint16_t shifted) {
    const uint8_t shift = MOD_BIT(KC_LSFT) | MOD_BIT(KC_RSFT);
    if (record->event.pressed) {
        bool shift_pressed = keyboard_report->mods & shift;
        const uint16_t target = shift_pressed ? shifted : normal;
        uint8_t keycode = target & 0xFF;
        if (keycode == KC_NO) {
            return;
        }
        press_key_with_level_mods(target);
    }
}
```

For this to work you don't need another layer, just define the marcors M(1) and M(2) on the normal default layer.

Then you would use it like this
```c
const macro_t *action_get_macro(keyrecord_t *record, uint8_t id, uint8_t opt) 
{
    switch(id) { 
        case 1: // M(1)
            override_key(KC_LBRACKET, KC_RBRACKET);
            break;  
        case 2: // M(2)
            override_key(UK_QUOT, KC_QUOT);
            break;  
```

The first parameter is the keycode to send for an unshifted press, while the second one is for shifted presses. The parameters can be any 16-bit keycode, no matter if it's shifted or not. It can also contain AltGr, and even AltGr-shift combinations. 

I haven't made a pull request for the above above yet, since I'm still thinking about the best way to do it, and the above doesn't work perfectly. I also need a better name for `override_key`

For example the key can't be held, and the key press will be sent immediately on key down. The reason for this is that it's impossible to send both a shifted and unshifted key at the same time to the operating system. So what the code instead does, is to enable the shift state required for the key, then send a press, followed by pressing all the old modifiers again, so that the next keypress works properly. It also takes special modifiers that are enabled by software, into account.

Anakey1, 2016-12-28:
ok so i would need that first large bit of code to define what a shift does? Then the smaller bit to define what to do on the macro depending on if a shift is present or not? 

so it would basically be like:

If shift and [ is pressed 
then output ]
else if [ only is pressed 
then output [

fredizzimo, 2016-12-28:
Yes, the first code block is just supporting functionality, which handles a lot of special cases and QMK internal stuff. As I said, similar functionality should ideally be provided by QMK, but for now you can just include the code snippet in your keymap.

The second on is the actual usage, and can be modified as you want. And yes the example should do what you show, if the rest of the keymap is modified accordingly and calls the right macro functions.

Anakey1, 2016-12-28:
Ok I added the cose to my keymap and i am getting errors when compiling. I have updated the keymap in the original post with the updated one with your code having taken out the extra layer.

fredizzimo, 2016-12-28:
I'm not sure exactly what's wrong without seeing the exact error message.

However you have at least forgot to paste the `press_key_with_level_mods` function to your keymap.

Anakey1, 2016-12-28:
so i just paste `press_key_with_level_mods` where it states function before the key layouts? This is my first time with keymaps and programminng etc so i will be making simple mistakes quite a bit

fredizzimo, 2016-12-28:
Yes, just paste

```c
void press_key_with_level_mods(uint16_t key) {
    const uint8_t interesting_mods = MOD_BIT(KC_LSHIFT) | MOD_BIT(KC_RSHIFT) | MOD_BIT(KC_RALT);

    // Save the state
    const uint8_t real_mods = get_mods();
    const uint8_t weak_mods = get_weak_mods();
    const uint8_t macro_mods = get_macro_mods();

    uint8_t target_mods = (key >> 8) & (QK_MODS_MAX >> 8);
    // The 5th bit indicates that it's a right hand mod,
    // which needs some fixup
    if (target_mods & 0x10) {
        target_mods &= 0xF;
        target_mods <<= 4;
    }

    // Clear the mods that we are potentially going to modify,
    del_mods(interesting_mods);
    del_weak_mods(interesting_mods);
    del_macro_mods(interesting_mods);

    // Enable the mods that we need
    add_mods(target_mods & interesting_mods);

    // Press and release the key
    register_code(key & 0xFF);
    unregister_code(key & 0xFF);

    // Restore the previous state
    set_mods(real_mods);
    set_weak_mods(weak_mods);
    set_macro_mods(macro_mods);
    send_keyboard_report();
}
```

Before line 141 containing `override_key`

Anakey1, 2016-12-28:
ah ok sorry i thought that was just the example of what the possibilities of the code could be rather then actually being part of the lower smaller bit of code. After posting in the large code i am still getting errors

http://i39.photobucket.com/albums/e177/Anakey1/9932c06d-da1f-4b31-a915-a3d9e3b30759_zps1m91xh19.png

Anakey1, 2016-12-28:
updated screenshot of errors as it was unclear

fredizzimo, 2016-12-28:
Ah sorry my fault

```c
       case 1: // M(1)
            override_key(KC_LBRACKET, KC_RBRACKET);
            break;  
        case 2: // M(2)
            override_key(UK_QUOT, KC_QUOT);
            break;  
```

should be

```c
       case 1: // M(1)
            override_key(record, KC_LBRACKET, KC_RBRACKET);
            break;  
        case 2: // M(2)
            override_key(record, UK_QUOT, KC_QUOT);
            break;  
```

Anakey1, 2016-12-28:
Excellent thanks a lot it works now :)

closed 2016-12-28 by Anakey1