qmk/qmk_firmware #967: Music mode and tap keys on layer 0 causes issues

ishtob, 2016-12-23:
I have 3 tap keys on the base layer (my qwerty layer). noticed the 3 tap keys will not play the note on the scale unless the tap key is held down or tapped twice. Also notice the first tap would result in the tap key assigned to be sent (my semi-colon that I use for a mouse layer when held for example would type ; during music mode)

jackhumbert, 2016-12-30:
I think this is the same issue that I mentioned in the subreddit, but I'll post my response here as well:

> This is a bug in the `process_music.c`implementation of limiting the keycodes that get passed through - currently all regular keycodes are blocked (`< 0xFF`), and ones above are let through. This is done so LOWER and RAISE still work (so you can turn music mode off again), but unfortunately things like LT exist in the same range :/ I'm not sure there's an easy fix right now, other than changing the value on line 138 to something like `0x7000`

drashna, 2018-10-21:
I think this handling was improved with the music mode mask function that @jackhumbert added "recently". 

And if that's the case, this can be closed, I think.

closed 2019-01-28 by ishtob