qmk/qmk_firmware #995: make infinity-<keymap>-dfu-util does not work.

mneme, 2017-01-08:
Output:
```Making ergodox/infinity with keymap mneme and target dfu-util

arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  33004	      0	  33004	   80ec	ergodox_infinity_mneme.hex

Size after:
   text	   data	    bss	    dec	    hex	filename
      0	  33004	      0	  33004	   80ec	ergodox_infinity_mneme.hex

dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Deducing device DFU version from functional descriptor length
dfu-util: More than one DFU capable USB device found! Try `--list' and specify the serial number or disconnect all but one device
```

Flashing normal kiihbod firmware to the Ergodox Infinity works just fine, but I can't seem to make qmk firmware to flash at all.

fredizzimo, 2017-01-08:
It seems like you have two dfu-util compatible devices connected and unfortunately the makefile only supports one device at the same time.

The easiest way is probably to disconnect the other device temporarily while flashing. You should be able to get a list of devices by running `dfu-util --list`. If you can't disconnect the other device, then you will have to flash it manually with `dfu-util --device xxx:xxx -D .build/ergodox_infinity_mneme.bin`. The bin file is automatically made when you run make `infinity-<keymap>-dfu-util`, even if it fails. I think it also can be generated by `make infinity-<keymap>-.build/ergodox_infinity_<keymap>.bin` but I haven't tested.

mneme, 2017-01-08:
Am I missing something?

```
mneme@adHominem ~/P/q/k/ergodox> dfu-util --list
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Deducing device DFU version from functional descriptor length
Found Runtime: [05ac:8289] ver=0131, devnum=13, cfg=1, intf=3, path="20-3.3", alt=0, name="UNKNOWN", serial="UNKNOWN"
Found DFU: [1c11:b007] ver=0000, devnum=15, cfg=1, intf=0, path="20-1", alt=0, name="Kiibohd DFU", serial="mk20dx256vlh7"
```

mneme, 2017-01-08:
As I said, flashing kiihbod firmware directly with dfu-util works just fine.

Is there a way to run make to just generate .bin files and not have it try to do it all. IE, so I can run dfu-util manually?

mneme, 2017-01-08:
And I am an idiot.

Thank you for pointing me to the bin-files.

Is there anywhere I can send you beer-money for the assistance?

fredizzimo, 2017-01-08:
First of all, as I said, I think that this command `make infinity-<keymap>-.build/ergodox_infinity_<keymap>.bin`, should generate the bin file, 

Secondly you can flash it manually, but notice that the .build folder is in the qmk root folder, not inside the Ergodox folder.

This command from the qmk root should work
`dfu-util --device 1c11:b007 -D .build/ergodox_infinity_mneme.bin`

fredizzimo, 2017-01-08:
Sorry, I didn't see your second message. 

And no, there's no need for any beer-money, although you wouldn't have to send it far, only to Helsinki. 

The documentation for this really has to be improved though, since you are not the first one to ask. I'm not sure where the most natural location would be though, since it affects several different keyboards, and there shouldn't be too many specific things in the main Ergodox readme.

closed 2017-02-07 by jackhumbert