qmk/qmk_firmware #997: XD60 issues with LED control

backstabbd88, 2017-01-09:
This is my first custom keyboard project, forgive me if this is a noob question, i have almost zero background with programming.

I just can't seem to make led controls work ( BL_INC,BL_DEC,BL_TOGG)

from what i have learned from the wiki page,
i just have to add "BACKLIGHT_ENABLE = yes" in the makefile.

Assign ports B5, B6, or B7 for #define BACKLIGHT_PIN in config.h for the PWM-control.

but by doing so disables a particular column on my keyboard, maybe because the following pins are already
assigned in the key matrix.

from default
config.h
#define MATRIX_ROW_PINS { D0, D1, D2, D3, D5 }
#define MATRIX_COL_PINS { F0, F1, E6, C7, C6, B6, D4, B1, B7, B5, B4, D7, D6, B3 }
https://github.com/jackhumbert/qmk_firmware/blob/master/keyboards/xd60/config.h

Is there a workaround for this? or maybe i'm just not following the guides correctly?

Wilba6582, 2017-01-09:
It seems unlikely that the PCB would be using the same pin for the switch matrix and backlight. Do you know what pin is assigned to backlight control?

Wilba6582, 2017-01-09:
Maybe @stanleylai can help identify which pins connect to the backlight?

xream, 2017-01-10:
I can offer you a hard way. 
You can try the left pins:
[image]

Wilba6582, 2017-01-10:
The current backlight code only works when `BACKLIGHT_PIN` is defined as B5, B6, or B7, because it's using built-in features of the ATmega to drive that pin with PWM.

It's not compatible with the pins being used on the XD60, which are defined here: 
https://github.com/jackhumbert/qmk_firmware/blob/master/keyboards/xd60/xd60.h

If you just want to turn on/off all the LEDs (no brightness control), you can call those functions.

backstabbd88, 2017-01-10:
@xream thank you, i'm afraid that will be too complicated for a beginner like me. I might just have to give up on the backlight led for now.

@Wilba6582 does the  B5, B6, or B7 limitation for PWM control only in qmk firmware? cause the stock firmware when i got the pcb had working backlight control for both brightness level and breathing. Thank you...

Wilba6582, 2017-01-11:
The backlight brightness code in QMK is what I originally wrote for the Planck, where I used one of the PWM pins of the Atmega32U4 to drive all the LEDs, thus using the Atmega's timer to produce the PWM output. I copied this for all the other OLKB boards, JD45, JD40 Mk II etc. and the code was refactored to use other pins.

The stock firmware might have been doing PWM in software instead... a fast timer with an interrupt and in the interrupt handler, it toggles the pin high/low.

There's a few PCBs now (GH60 clones etc.) that use QMK so it's probably worth the effort to generalize the backlight code a little more so it will work on one or more pins. I've already done a fix for someone else's NeRD 60 that uses QMK, where the switch and side LEDs use one pin each. I'm also helping someone else with their JD45, he wants the backlight to do different things depending on what layer is active.

Maybe @jackhumbert can start a discussion about this, to get all the ideas in the one place?

xream, 2017-01-11:
@Wilba6582 I think you're working on something like this: https://github.com/kairyu/tmk_keyboard_custom/blob/master/keyboard/gh60/backlight.c . Is there any working example that I can learn from? Thank you.

Wilba6582, 2017-01-11:
Yeah that's basically what I was talking about. Code like that could be refactored into the core QMK backlight handling.

fredizzimo, 2017-01-16:
I tried to do some Google search, but I couldn't find out what the default firmware for the keyboard is? Is it something completely custom made, or something based on another open source firmware?

backstabbd88, 2017-01-17:
@fredizzimo  

I'm pretty sure it uses tmk as default.

https://world.taobao.com/item/537485105086.htm?spm=a312a.7700714.0.0.oRXd3B#detail
translated description - "3. Can use KLE custom key bits, support TKG-TOOLS offline brush machine and TKG online brush machine"

I also tried flashing tmk firmware with https://tkg.io/ and can confirm that backlight controls works with it.

fredizzimo, 2017-01-17:
In that case the source code for the default fimrware should be [here]( https://github.com/kairyu/tmk_keyboard_custom/tree/master/keyboard/gh60) (which I now see is already linked by @xream above).

Although the softpwm implementation seems to be enabled by default, the ifdefs indicate that hardware PWM on port `B6` is possible. For the backlight to work, at least at its full potential, you probably need the led_matrix code as well.

Another thing that I noticed is that the matrix is defined differently for `GH60_REV_CNY` [here]( https://github.com/kairyu/tmk_keyboard_custom/blob/master/keyboard/gh60/matrix.c). It's using `B7` instead of `B6` for the sixth column. But I have no idea why it works by default in QMK in that case, but maybe it really doesn't and certain NKRO combinations don't. 

So based on this you could try changing the matrix pin to B7 and set the BACKLIGHT_PIN to B6

backstabbd88, 2017-01-17:
row matrix  for 
qmk-xd60
F0, F1, E6, C7, C6, B6, D4, B1, B7, B5, B4, D7, D6, B3
 
tmk-gh60
F1  F0  E6  D7  D6  D4  C7  C6  B6  B5  B4  B3  B1  B0  (Rev.A)
F1  F0  E6  D7  D6  D4  C7  C6  B7  B6  B5  B4  B3  B1  (Rev.B)
F1  F0  E6  D7  D6  D4  C7  C6  B7  B5  B4  B3  B1  B0  (Rev.CHN/CNY)

hmm. maybe i could try it, http://qmk.sized.io/ has easy way to change the pins.

Problem is i just run into another issue, after switching to tmk, i could no longer flash my qmk hex with qmk flasher/ atmel flip. I'm using windows, and qmk flasher and atmel flip can't detect when i put my keyboard to bootloader mode. I already tried reinstalling drivers.

fredizzimo, 2017-01-17:
Ah there seems to be many other differences as well, I just checked that B6 wasn't there. It seems very strange that the pins are so different. 

Unfortunately I'm not familiar enough with the flashing tools to help you with that. But did you use the dfu bootloader when flashing through TKG? The other ones are probably not compatible.

backstabbd88, 2017-01-17:
I used  qmk firmware flasher before. https://github.com/jackhumbert/qmk_firmware_flasher/releases

Tried to switch to tmk and succesfully flashed using https://github.com/kairyu/tkg-toolkit.

But now i can no longer flash back to qmk with the qmk flasher. The app cant seem to detect my keyboard when i put it to dfu bootloader mode.

closed 2017-06-27 by skullydazed